bulletphysics/bullet3 #2189: pybullet crashes when used in Blender scripts

JayFoxRox, 2019-04-07:
I'm unable to use pybullet in Blender scripts.

I'm running:

- Arch Linux (x64)
- pybullet 2.4.8 (latest on PIP)
- Python 3.7.3 (latest on Arch Repos)
- Blender 17:2.79.b.git5.a29446da-2 (latest on Arch Repos)
- Blender 2.80.190326.adfdae3fc2f4-1 (latest on Arch User Repos)

To reproduce, just open a scripting area in Blender, and type:

```python
>>> import pybullet
>>> i = pybullet.connect(pybullet.DIRECT)
>>> pybullet.resetSimulation(i)
```

The last command will reliably crash Blender 2.79 (and the latest Blender 2.80 builds).
The same set of commands work fine in stand-alone Python on the same machine.

Some other commands also result in crashes.

---

The crash I'm getting has the following backtrace:

```
$ gdb --args blender --python-expr "import pybullet; i = pybullet.connect(pybullet.DIRECT); pybullet.resetSimulation(i)"
GNU gdb (GDB) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from blender...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/bin/blender --python-expr import\ pybullet\;\ i\ =\ pybullet.connect\(pybullet.DIRECT\)\;\ pybullet.resetSimulation\(i\)
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffe39ff700 (LWP 6185)]
[New Thread 0x7fffe24fe700 (LWP 6186)]
[New Thread 0x7fffe1cfd700 (LWP 6187)]
[New Thread 0x7fffe14fc700 (LWP 6188)]
[New Thread 0x7fffe0cfb700 (LWP 6189)]
Read prefs: /home/fox/.config/blender/2.79/config/userpref.blend
[New Thread 0x7fffdf6c2700 (LWP 6190)]
[Detaching after fork from child process 6191]
[Thread 0x7fffdf6c2700 (LWP 6190) exited]
[New Thread 0x7fffdf6c2700 (LWP 6192)]
[New Thread 0x7fffde9ff700 (LWP 6193)]
[Thread 0x7fffdf6c2700 (LWP 6192) exited]
[New Thread 0x7fffdf6c2700 (LWP 6194)]
[New Thread 0x7fffce1fe700 (LWP 6195)]
[New Thread 0x7fffccdff700 (LWP 6196)]
[New Thread 0x7fffcc5fe700 (LWP 6197)]
[New Thread 0x7fffcc3fd700 (LWP 6198)]
[New Thread 0x7fffcb5ff700 (LWP 6199)]
[Thread 0x7fffcc3fd700 (LWP 6198) exited]
[New Thread 0x7fffc9f4d700 (LWP 6200)]
pybullet build time: Mar 26 2019 10:41:25

Thread 1 "blender" received signal SIGSEGV, Segmentation fault.
0x0000555556ed3370 in btDiscreteDynamicsWorld::~btDiscreteDynamicsWorld() ()
(gdb) bt
#0  0x0000555556ed3370 in btDiscreteDynamicsWorld::~btDiscreteDynamicsWorld() ()
#1  0x00007fffc5000b4a in btSoftMultiBodyDynamicsWorld::~btSoftMultiBodyDynamicsWorld (this=0x7fffc6665e10, __in_chrg=<optimized out>) at src/BulletSoftBody/btSoftMultiBodyDynamicsWorld.cpp:60
#2  0x00007fffc4e1f13e in PhysicsServerCommandProcessor::deleteDynamicsWorld (this=0x7fffc9686420) at examples/SharedMemory/PhysicsServerCommandProcessor.cpp:2834
#3  0x00007fffc4e333c5 in PhysicsServerCommandProcessor::resetSimulation (this=this@entry=0x7fffc9686420) at examples/SharedMemory/PhysicsServerCommandProcessor.cpp:11835
#4  0x00007fffc4e3377f in PhysicsServerCommandProcessor::processResetSimulationCommand (this=0x7fffc9686420, clientCmd=..., serverStatusOut=..., bufferServerToClient=<optimized out>, bufferSizeInBytes=<optimized out>) at examples/SharedMemory/PhysicsServerCommandProcessor.cpp:8603
#5  0x00007fffc4e45772 in PhysicsServerCommandProcessor::processCommand (this=0x7fffc9686420, clientCmd=..., serverStatusOut=..., bufferServerToClient=0x7fffc11c0d70 "SDNANAME{\001", bufferSizeInBytes=<optimized out>) at examples/SharedMemory/PhysicsServerCommandProcessor.cpp:11207
#6  0x00007fffc4e1b2a8 in PhysicsDirect::submitClientCommand (command=..., this=0x7fffc9686450) at examples/SharedMemory/PhysicsDirect.cpp:1192
#7  PhysicsDirect::submitClientCommand (this=0x7fffc9686450, command=...) at examples/SharedMemory/PhysicsDirect.cpp:1167
#8  0x00007fffc4e71614 in b3SubmitClientCommand (commandHandle=<optimized out>, physClient=<optimized out>) at examples/SharedMemory/PhysicsClientC_API.cpp:2430
#9  b3SubmitClientCommand (physClient=<optimized out>, commandHandle=<optimized out>) at examples/SharedMemory/PhysicsClientC_API.cpp:2422
#10 0x00007fffc4e716b4 in b3SubmitClientCommandAndWaitStatus (physClient=physClient@entry=0x7fffc9686450, commandHandle=0x7fffc11aea68) at examples/SharedMemory/PhysicsClientC_API.cpp:2454
#11 0x00007fffc4cbf08a in pybullet_resetSimulation (self=<optimized out>, args=<optimized out>, keywds=<optimized out>) at examples/pybullet/pybullet.c:1957
#12 0x00007ffff75b1fa4 in _PyMethodDef_RawFastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#13 0x00007ffff75b2101 in _PyCFunction_FastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#14 0x00007ffff7622d19 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.7m.so.1.0
#15 0x00007ffff756ad09 in _PyEval_EvalCodeWithName () from /usr/lib/libpython3.7m.so.1.0
#16 0x00007ffff756bc64 in PyEval_EvalCodeEx () from /usr/lib/libpython3.7m.so.1.0
#17 0x00007ffff756bc8c in PyEval_EvalCode () from /usr/lib/libpython3.7m.so.1.0
#18 0x00007ffff7694694 in ?? () from /usr/lib/libpython3.7m.so.1.0
#19 0x00007ffff769569d in PyRun_StringFlags () from /usr/lib/libpython3.7m.so.1.0
#20 0x0000555556543871 in BPY_execute_string_ex ()
#21 0x000055555611a3b8 in arg_handle_python_expr_run ()
#22 0x0000555556b71f67 in BLI_argsParse ()
#23 0x0000555556103172 in main ()
(gdb) 
```

Here's more information about the Blender:

```
$ blender --version
Blender 2.79 (sub 7)
	build date: 2019-03-27
	build time: 18:12:23
	build commit date: 2019-03-26
	build commit time: 13:42
	build hash: a29446da526d
	build platform: Linux
	build type: Release
	build c flags:  -Wall -Wcast-align -Werror=implicit-function-declaration -Werror=return-type -Werror=vla -Wstrict-prototypes -Wmissing-prototypes -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wlogical-op -Wundef -Winit-self -Wnonnull -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Wformat-signedness -Wrestrict -Wuninitialized -Wredundant-decls -Wshadow -Wno-error=unused-but-set-variable -Wimplicit-fallthrough=5 -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -fuse-ld=gold -fopenmp -std=gnu99   -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -msse2
	build c++ flags:  -Wredundant-decls -Wall -Wno-invalid-offsetof -Wno-sign-compare -Wlogical-op -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Werror=return-type -Werror=implicit-function-declaration -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wundef -Wformat-signedness -Wrestrict -Wuninitialized -Wundef -Wmissing-declarations -Wimplicit-fallthrough=5 -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -fuse-ld=gold -fopenmp -std=c++11   -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -msse2
	build link flags:  -Wl,--version-script='/build/blender/src/blender-2.79.b.git5.a29446da/source/creator/blender.map'
	build system: CMake
```

*I'm reporting this as pybullet issue, because it crashes in pybullet code (inside of Blender). I assume this could have to do with Bullet being used in Blender internally, and possible library version conflicts or context mixups. If this turns out to be a problem in Blender, I'll move this issue to the respective bugtracker.*

erwincoumans, 2019-04-07:
It is likely because you compile PyBullet using a different Python library (system) than Blender own distributed Python version.

closed 2019-04-07 by erwincoumans

JayFoxRox, 2019-04-07:
Thanks for taking the time to respond.

> It is likely because you compile PyBullet using a different Python library (system) than Blender own distributed Python version

Blender uses the system version of Python, as can be seen in the backtrace:

```
[...] from /usr/lib/libpython3.7m.so.1.0
```

To confirm this, here is the version info from within Blender:

```
PYTHON INTERACTIVE CONSOLE 3.7.3 (default, Mar 26 2019, 21:43:19)  [GCC 8.2.1 20181127]
```

And system Python:

```
Python 3.7.3 (default, Mar 26 2019, 21:43:19) 
[GCC 8.2.1 20181127] on linux
```

\- in fact, this is standard on Arch Linux (Blender 2.79 uses the system installation of Python).
For the Blender-2.8-bin AUR package (which ships its own Python) I've contacted the maintainer to move towards system Python (and did local patches as that patch hasn't been integrated in AUR).

All three versions of Blender crash:

- 2.79 / system Python
- 2.8 / system Python
- 2.8 / packaged Python

At this point, I only have 1 Python installation on this system; it's the one used by my Python shell + Blender 2.79 + Blender 2.8 + the one used to install PyBullet through pip.

> Also, Blender uses a different Bullet version thay PyBullet, that might also interfere.

Yes, that's what I also think; I'd consider that a bug in Bullet.

If the symbols are incompatible, they should be versioned differently.
PyBullet should also load it's own symbols by explicitly specifying which library it wants to use.

The Bullet API is designed around handling separate connections, it shouldn't be a problem to have different connections on different library versions. A connection made by PyBullet should be independent from those made by the C++ API or another Python environment (in the same process).
The context of each PyBullet import should be isolated.

If my program (be it Blender or something else) crashes because I attempt to use PyBullet, then it's an issue that should be solved (even if it only meant sanity checks and throwing an exception for now).

> Either way, this is a Blender issue, not Bullet/PyBullet.

With the reasoning presented in this post, I'd assume that this is a PyBullet issue.

However, I don't even know why it would crash (yet). The whole "library conflicts" is just a theory.
So it would be useful if someone looked into it.

If you insist, I'll move this to the Blender bugtracker.
However, I assume they'll redirect me here, as the crash is within PyBullet libraries.

> It runs fine if you compile PyBullet with the Blender version of Python.

This is not the issue.

As stated above, Blender uses my native systems Python.

It can still be the Bullet in Blender vs. Bullet in PyBullet conflict.
However, that's something that shouldn't happen either: it would be a bug in PyBullet dynamic linking.