bulletphysics/bullet3 #2194: Problems about getDebugVisualizerCamera()

pyni, 2019-04-09:
Hi, i want to reset the return parameters (width, height, viewMat, projMat, cameraUp, camForward, horizon,vertical, _,_,dist, camTarget) of **p.getDebugVisualizerCamera()** in order to change the view direction of point clouds in file: **pointCloudFromCameraImage.py**.
when I run these codes:
    **base_pos,orn = p.getBasePositionAndOrientation(trayID)
    view_matrix = p.computeViewMatrixFromYawPitchRoll(cameraTargetPosition=base_pos,distance=1.3,yaw=180,pitch=-90,roll=0,upAxisIndex=2)
    proj_matrix =  p.computeProjectionMatrixFOV(fov=60, aspect=float(960)/720,nearVal=0.1, farVal=100.0)
    (_, _, px, _, _) =  p.getCameraImage(
        width=960, height=720, viewMatrix=view_matrix,
        projectionMatrix=proj_matrix, renderer= p.ER_BULLET_HARDWARE_OPENGL)**
it shows:
[image]

Then i run following order to reset the parameters

pyni, 2019-04-09:
Actually , i do not know how to change **horizon** and **vertical** of function **getDebugVisualizerCamera** in file **pointCloudFromCameraImage.py**:
[image]
[image]
I think it is important for point cloud generation

pyni, 2019-04-09:
Hi, I find if i set **p.resetDebugVisualizerCamera(1.3,180,-89,base_pos)**,
the parameters seem right:
**(1024, 768, (-0.9999999403953552, 8.740908441495776e-08, -1.5257376473343243e-09, 0.0, -8.74223999858259e-08, -0.9998475909233093, 0.017452478408813477, 0.0, 0.0, 0.017452476546168327, 0.9998476505279541, 0.0, -0.0, 1.862645149230957e-09, -1.2999999523162842, 1.0), (0.7499999403953552, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, -1.0000200271606445, -1.0, 0.0, 0.0, -0.02000020071864128, 0.0), (0.0, 0.0, 1.0), (1.5257376473343243e-09, -0.017452478408813477, -0.9998476505279541), (-26666.66796875, -0.002331264317035675, 0.0), (0.0017481818795204163, -19996.953125, 349.049560546875), 180.0, -89.0, 1.2999999523162842, (0.0, 0.0, 0.0))**

erwincoumans, 2019-04-14:
The debug visualizer camera cannot be set to look straight down at a pitch of -90, due to Euler angle singularity.

closed 2019-04-14 by erwincoumans

Flymek, 2020-02-05:
Hi, could you then explain the horizon and vertical variables? How can I calculate them myself to be able to take a point cloud looking straight down? I sadly cannot find that information anywhere, but I know all the other variables debug visualizer camera is supposed to give me.