bulletphysics/bullet3 #2196: A problem about point cloud

pyni, 2019-04-09:
HI, i use **pointCloudFromCameraImage.py** to generate point clouds. 
However, i find some points are under the object!! This is very strange:
[image]
[image]

Could you please provide with some suggestions?

pyni, 2019-04-10:
[image]
[image]
Actually,  i can hardly understand why the depth data of these two yellow objects are total different.

erwincoumans, 2019-04-10:
Please share a small reproduction case, otherwise no-one can help you (attach a zipfile or link to some github repo etc).

pyni, 2019-04-10:
Hi, i have found the problem.

It seems that pybullet can not support 'stl' file which normals are pointing inside of the surface.

closed 2019-04-10 by pyni