bulletphysics/bullet3 #2207: Video output of joints SDF v/s URDF

Prakhar005, 2019-04-16:
Hello,
I am creating a simple bot which has two square shaped links connected by a revolute joint. When I model the robot in the URDF format, the pybullet simulation output seems to give a more natural hinge motion of the joint as compared to the case when I model the robot using SDF format. I was wondering if this difference is related to the manner in which PyBullet processes the two formats and if there is anything I could do to fix this?
Thanks!

erwincoumans, 2019-04-16:
The SDF and URDF implementation is shared. Attach a zipped version of both, so someone can look at it.

closed 2019-04-16 by erwincoumans

Prakhar005, 2019-04-16:
[bot_two_links.zip](https://github.com/bulletphysics/bullet3/files/3085699/bot_two_links.zip)
Hello,
Thank you for your reply. I have attached the zip file which contains the SDF and the URDF models for the robot. Please let me know if anything else is needed. Thank you very much.