bulletphysics/bullet3 #2212: why stepSimulation of b3RobotSimulatorClientAPI_NoDirect running slowly?

Evi5, 2019-04-18:
stepSimulation of **b3RobotSimulatorClientAPI_NoDirect** running slowly

stepSimulation of **btDiscreteDynamicsWorld** running so fast!

Why is that?
And is there some diff of the two stepSimulation?

I want to use urdf in c with b3RobotSimulatorClientAPI, but the stepSimulation of it is too slowly (>200times).

Macos 10.14.3

erwincoumans, 2019-04-18:
What timestep and number of solver iterations did you use? Default in robotics is 240 Hz and 50 iterations. How did you benchmark exactly?

Evi5, 2019-04-19:
changed the same program for diff with stepSimulation (b3RobotSimulatorClientAPI_NoDirect or btDiscreteDynamicsWorld)

debug results shows that  **slowly problem is caused by stepSimulation**

program A
`
        gContactProcessedCallback = myContactProcessedCallback;

        m_collisionConfiguration = new btDefaultCollisionConfiguration();

        m_dispatcher = new btCollisionDispatcher(m_collisionConfiguration);

        btVector3 worldAabbMin(-10000, -10000, -10000);
        btVector3 worldAabbMax(10000, 10000, 10000);
        m_broadphase = new btAxisSweep3(worldAabbMin, worldAabbMax);

        m_solver = new btSequentialImpulseConstraintSolver;

        m_dynamicsWorld = new btDiscreteDynamicsWorld(m_dispatcher, m_broadphase, m_solver, m_collisionConfiguration);

        btScalar timeStep = 0.01;
        m_dynamicsWorld->stepSimulation(timeStep, 50, timeStep);
`

program B
`

        b3RobotSimulatorClientAPI m_robotSim;

        m_robotSim.stepSimulation();
`

total running time:
program A  0.837053 seconds
program B  89.4724 seconds

the program is try to train 10 quadrupedRobots to walk in 1 time, 

I am not sure the solver iterations, maybe it's the key? the flow is the source

hw_6_code_OpenGL_**0.837053 seconds**.zip:
        https://pan.baidu.com/s/1OtSu5VcWq2TrsLhctjYZDA
hw_6_code_RobotSimulatorClientAPI_**89.4724 seconds**.zip
        https://pan.baidu.com/s/1luIqDR2fpvxD8tgaQmFzHQ

Sim_Simulator.hpp is the important codes.

erwincoumans, 2019-04-19:
Timestep should be 1./240. to be equal.
m_dynamicsWorld->getSolverInfo().m_numSolverIterations=50
What is inside each world? b3RobotSimulatorClientAPI creates btMultiBody by default.

Evi5, 2019-04-19:
there is no m_numSolverIterations but m_numIterations

program A create an robot by c codes, and add to world
program B load an robot by urdf, and add to word

I changed program A to 

    m_dynamicsWorld->getSolverInfo().m_numIterations=50;
    btScalar timeStep = 1./240.;
    m_dynamicsWorld->stepSimulation(timeStep, 50, timeStep);

total time up to 2.05036 seconds, also much fast than 89.4724 seconds.

download link has been delete.
Sim_Simulator.hpp is the important codes.

codes
[hw_6_code_RobotSimulatorClientAPI_89.4724 seconds.zip](https://github.com/bulletphysics/bullet3/files/3097100/hw_6_code_RobotSimulatorClientAPI_89.4724.seconds.zip)
[hw_6_code_OpenGL_2.05036 seconds.zip](https://github.com/bulletphysics/bullet3/files/3097115/hw_6_code_OpenGL_2.05036.seconds.zip)

closed 2019-04-19 by Evi5