bulletphysics/bullet3 #2213: pybullet renders wrong color

OhmySallyGarden, 2019-04-19:
I load the URDF model of kinova jaco, and I find that the color of the robot in the simulation is not corresponding to the camera image. 
[image]
upper left of the image and the environment are not same.
The urdf description is : 

<?xml version="1.0" ?>

<robot name="j2n6s300" xmlns:body="http://playerstage.sourceforge.net/gazebo/xmlschema/#body" xmlns:controller="http://playerstage.sourceforge.net/gazebo/xmlschema/#controller" xmlns:gazebo="http://playerstage.sourceforge.net/gazebo/xmlschema/#gz" xmlns:geom="http://playerstage.sourceforge.net/gazebo/xmlschema/#geom" xmlns:interface="http://playerstage.sourceforge.net/gazebo/xmlschema/#interface" xmlns:joint="http://playerstage.sourceforge.net/gazebo/xmlschema/#joint" xmlns:model="http://playerstage.sourceforge.net/gazebo/xmlschema/#model" xmlns:physics="http://playerstage.sourceforge.net/gazebo/xmlschema/#physics" xmlns:renderable="http://playerstage.sourceforge.net/gazebo/xmlschema/#renderable" xmlns:rendering="http://playerstage.sourceforge.net/gazebo/xmlschema/#rendering" xmlns:sensor="http://playerstage.sourceforge.net/gazebo/xmlschema/#sensor" xmlns:xacro="http://ros.org/wiki/xacro" xmlns:xi="http://www.w3.org/2001/XInclude">
  
  
  <link name="root">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0 0 0"/>
      </geometry>
      
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0 0 0"/>
      </geometry>
    </collision>
  </link>
  
  <link name="world"/>
  <joint name="connect_root_and_world" type="fixed">
    <child link="root"/>
    <parent link="world"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
  
  <gazebo>
    <plugin filename="libgazebo_ros_control.so" name="gazebo_ros_control">
      <robotNamespace>j2n6s300</robotNamespace>
      <robotSimType>gazebo_ros_control/DefaultRobotHWSim</robotSimType>
    </plugin>
  </gazebo>
  <link name="j2n6s300_link_base">
    <visual>
      <geometry>
        <mesh filename="package:///home/hiter/Git/DDPG/j2n6s300_model/meshes/base.dae"/>
      </geometry>
      <material name="carbon_fiber">
        <color rgba="0.1788232 0.1788232 0.1788232 1"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package:///home/hiter/Git/DDPG/j2n6s300_model/meshes/base.dae"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.46784"/>
      <origin rpy="0 0 0" xyz="0 0 0.1255"/>
      <inertia ixx="0.000951270861568" ixy="0" ixz="0" iyy="0.000951270861568" iyz="0" izz="0.000374272"/>
    </inertial>
  </link>
  <joint name="j2n6s300_joint_base" type="fixed">
    <parent link="root"/>
    <child link="j2n6s300_link_base"/>
    <axis xyz="0 0 0"/>
    <limit effort="2000" lower="0" upper="0" velocity="1"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <dynamics damping="0.0" friction="0.0"/>
  </joint>
  <link name="j2n6s300_link_1">
    <visual>
      <geometry>
[... 448 lines cut ...]
        <mesh filename="package:///home/hiter/Git/DDPG/j2n6s300_model/meshes/finger_distal.dae"/>
      </geometry>
      <material name="carbon_fiber1">
        <color rgba="0.5960783 0.5960783 0.5960783 1"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package:///home/hiter/Git/DDPG/j2n6s300_model/meshes/finger_distal.dae"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.01"/>
      <origin xyz="0.022 0 0"/>
      <inertia ixx="7.8999684e-07" ixy="0" ixz="0" iyy="7.8999684e-07" iyz="0" izz="8e-08"/>
    </inertial>
  </link>
  <joint name="j2n6s300_joint_finger_tip_3" type="fixed">
    <parent link="j2n6s300_link_finger_3"/>
    <child link="j2n6s300_link_finger_tip_3"/>
    <axis xyz="0 0 1"/>
    <origin rpy="0 0 0" xyz="0.044 -0.003 0"/>
    <limit effort="2000" lower="0" upper="2" velocity="1"/>
  </joint>
</robot>

OhmySallyGarden, 2019-04-19:
`
<?xml version="1.0" ?>

<robot name="j2n6s300" xmlns:body="http://playerstage.sourceforge.net/gazebo/xmlschema/#body" xmlns:controller="http://playerstage.sourceforge.net/gazebo/xmlschema/#controller" xmlns:gazebo="http://playerstage.sourceforge.net/gazebo/xmlschema/#gz" xmlns:geom="http://playerstage.sourceforge.net/gazebo/xmlschema/#geom" xmlns:interface="http://playerstage.sourceforge.net/gazebo/xmlschema/#interface" xmlns:joint="http://playerstage.sourceforge.net/gazebo/xmlschema/#joint" xmlns:model="http://playerstage.sourceforge.net/gazebo/xmlschema/#model" xmlns:physics="http://playerstage.sourceforge.net/gazebo/xmlschema/#physics" xmlns:renderable="http://playerstage.sourceforge.net/gazebo/xmlschema/#renderable" xmlns:rendering="http://playerstage.sourceforge.net/gazebo/xmlschema/#rendering" xmlns:sensor="http://playerstage.sourceforge.net/gazebo/xmlschema/#sensor" xmlns:xacro="http://ros.org/wiki/xacro" xmlns:xi="http://www.w3.org/2001/XInclude">
  
  
  <link name="root">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0 0 0"/>
      </geometry>
      
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0 0 0"/>
      </geometry>
    </collision>
  </link>
  
  <link name="world"/>
  <joint name="connect_root_and_world" type="fixed">
    <child link="root"/>
    <parent link="world"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
  
  <gazebo>
    <plugin filename="libgazebo_ros_control.so" name="gazebo_ros_control">
      <robotNamespace>j2n6s300</robotNamespace>
      <robotSimType>gazebo_ros_control/DefaultRobotHWSim</robotSimType>
    </plugin>
  </gazebo>
  <link name="j2n6s300_link_base">
    <visual>
      <geometry>
        <mesh filename="package:///home/hiter/Git/DDPG/j2n6s300_model/meshes/base.dae"/>
      </geometry>
      <material name="carbon_fiber">
        <color rgba="0.1788232 0.1788232 0.1788232 1"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package:///home/hiter/Git/DDPG/j2n6s300_model/meshes/base.dae"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.46784"/>
      <origin rpy="0 0 0" xyz="0 0 0.1255"/>
      <inertia ixx="0.000951270861568" ixy="0" ixz="0" iyy="0.000951270861568" iyz="0" izz="0.000374272"/>
    </inertial>
  </link>
  <joint name="j2n6s300_joint_base" type="fixed">
    <parent link="root"/>
    <child link="j2n6s300_link_base"/>
    <axis xyz="0 0 0"/>
    <limit effort="2000" lower="0" upper="0" velocity="1"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <dynamics damping="0.0" friction="0.0"/>
  </joint>
  <link name="j2n6s300_link_1">
    <visual>
      <geometry>
        <mesh filename="package:///home/hiter/Git/DDPG/j2n6s300_model/meshes/shoulder.dae"/>
      </geometry>
      <material name="carbon_fiber">
        <color rgba="0.1788232 0.1788232 0.1788232 1"/>
[... 445 lines cut ...]
      </geometry>
      <material name="carbon_fiber1">
        <color rgba="0.5960783 0.5960783 0.5960783 1"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package:///home/hiter/Git/DDPG/j2n6s300_model/meshes/finger_distal.dae"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.01"/>
      <origin xyz="0.022 0 0"/>
      <inertia ixx="7.8999684e-07" ixy="0" ixz="0" iyy="7.8999684e-07" iyz="0" izz="8e-08"/>
    </inertial>
  </link>
  <joint name="j2n6s300_joint_finger_tip_3" type="fixed">
    <parent link="j2n6s300_link_finger_3"/>
    <child link="j2n6s300_link_finger_tip_3"/>
    <axis xyz="0 0 1"/>
    <origin rpy="0 0 0" xyz="0.044 -0.003 0"/>
    <limit effort="2000" lower="0" upper="2" velocity="1"/>
  </joint>
</robot>
`

OhmySallyGarden, 2019-04-19:
The upper left of the image is what I want the physics engine to render. And I do not know how  to use  getVisualShapeData() and changeVisualShape() to only chage the ring of the arm rather than the whole link.

maedmaex, 2019-08-02:
Hey, could you share any code to be able to help you

closed 2021-04-26 by erwincoumans