bulletphysics/bullet3 #2214: Modelling non-mechanical structures

Prakhar005, 2019-04-22:
Hello,
I am trying to model a piezoelectric sheet which expands when we apply a voltage signal to it. I was wondering if PyBullet allows the capability to simulate such structures because as far as I know we can only control the motors in this tool? Is there any other toll which could help in this direction? Thank you!

erwincoumans, 2019-05-31:
How does the element expand?

You could model it using an articulated btMultiBody, with motors.

If different deformation is important, you could use a convex hull with vertices, and modify the vertex locations. You will need to flush the contact caches and use a small timestep.

Alternatively try using a btSoftBody.

Prakhar005, 2019-06-03:
Hello,
Thank you for your reply. The sheet expands something like this : The sheet is lying flat on a surface and when a voltage is applied to it, it curves up like an arc. What would be an appropriate method to simulate this.

closed 2021-04-26 by erwincoumans