bulletphysics/bullet3 #2216: Memory leak in loadTexture

iory, 2019-04-23:
I have encountered a memory leak by using `pybullet.loadTexture`.
Following is the example of memory leak case.

```
import os

import numpy as np
import pybullet as p
import cv2

p.connect(p.DIRECT)

@profile
def test():
    texture_ids = []
    for i in range(1000):
        img = np.array(255 * np.random.random(size=(600, 600, 3)),
                       dtype=np.uint8)
        save_filename = './hoge{}.png'.format(i)
        cv2.imwrite(save_filename, img)
        texture_id = p.loadTexture(save_filename)
        texture_ids.append(texture_id)
        os.remove(save_filename)
    p.resetSimulation()
    print('after reset')

test()
```

I tested the avobe code by the following command.

```
python -m memory_profiler memory_leak.py
```

the profiling result is here.
```
Line #    Mem usage    Increment   Line Contents
================================================
    10  105.535 MiB  105.535 MiB   @profile
    11                             def test():
    12  105.535 MiB    0.000 MiB       texture_ids = []
    13 1242.316 MiB    0.000 MiB       for i in range(1000):
    14 1245.410 MiB    9.641 MiB           img = np.array(255 * np.random.random(size=(600, 600, 3)),
    15 1242.316 MiB    0.062 MiB                          dtype=np.uint8)
    16 1242.316 MiB    0.000 MiB           save_filename = './hoge{}.png'.format(i)
    17 1242.316 MiB    0.789 MiB           cv2.imwrite(save_filename, img)
    18 1242.316 MiB    4.102 MiB           texture_id = p.loadTexture(save_filename)
    19 1242.316 MiB    0.000 MiB           texture_ids.append(texture_id)
    20 1242.316 MiB    0.000 MiB           os.remove(save_filename)
    21 1232.379 MiB    0.000 MiB       p.resetSimulation()
    22 1232.379 MiB    0.000 MiB       print('after reset')
```

I examined this memory leak, and figure out the tinyrenderer plugin not release allocated memory correctly.
here https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/plugins/tinyRendererPlugin/TinyRendererVisualShapeConverter.cpp#L1245

erwincoumansepic, 2019-04-23:
Thanks for the report, but why did you close it? Did you try disabling file caching?

iory, 2019-04-24:
Sorry, I think that this issue is caused by my misunderstanding and I re-examined this issue still exists. So reopen this issue.

I add commit to do print debug, here https://github.com/iory/bullet3/blob/test/examples/SharedMemory/plugins/tinyRendererPlugin/TinyRendererVisualShapeConverter.cpp#L1241-L1251 .
also I checked by `strace` command, but it seems that the program not called `free` correctly.

Following is the results of `strace` command.
```
brk(0x55bd43f61000)                     = 0x55bd43f61000
brk(0x55bd44161000)                     = 0x55bd44161000
unlink("./hoge0.png")                   = 0
munmap(0x7f2d1756a000, 1081344)         = 0
write(1, "free address 0x55bd43c40840\n", 28free address 0x55bd43c40840
) = 28
write(1, "\337\n", 2
                    )                   = 2
write(1, "free 0 called\n", 14free 0 called
)         = 14
munmap(0x7f2d19526000, 1052672)         = 0
munmap(0x7f2d19627000, 1150976)         = 0
munmap(0x7f2d19740000, 6295552)         = 0
```

erwincoumansepic, 2019-04-25:
There are 2 issues:

1) PyBullet uses file caching by default: files are caches and not deleted after a 'resetSimulation'. You can disable this by adding this line after connect:
~~~~~~~~~~~
p.setPhysicsEngineParameter(enableFileCaching=0)
~~~~~~~~~~~
So the script could be:
~~~~~~~~~~~
import os
  
import numpy as np
import pybullet as p
import cv2

p.connect(p.DIRECT)
p.setPhysicsEngineParameter(enableFileCaching=0)

@profile
def test():
    texture_ids = []
    for i in range(300):
        img = np.array(255 * np.random.random(size=(600, 600, 3)),
                       dtype=np.uint8)
        save_filename = './hoge{}.png'.format(i)
        cv2.imwrite(save_filename, img)
        texture_id = p.loadTexture(save_filename)
        print("texture_id=",texture_id)
        texture_ids.append(texture_id)
        os.remove(save_filename)
    p.resetSimulation()
    print('after reset')
    p.disconnect()
    print('after disconnect')

test()

~~~~~~~~~~~

2) Even if file caching is disabled, some file loading data is not released until you disconnect
Either add a 'disconnect' and reconnect or wait for the following patch

https://github.com/bulletphysics/bullet3/pull/2208
This will be included in PyBullet 2.4.9 soon.

iory, 2019-04-25:
Thank you for your solution! 

I checked #2208 and it works correctly releasing memory.

closed 2019-04-25 by iory