bulletphysics/bullet3 #2219: build script should build a shared library of the C API

goretkin, 2019-04-28:
The build script produces a pybullet shared library. Could it also build a shared library of just the C API (`bullet3/examples/SharedMemory/PhysicsClientC_API.h`)?

erwincoumans, 2019-04-29:
That would be easy. What would you do with it?

goretkin, 2019-04-29:
I made an attempt at a PR that would solve this for my case.
I would like to use Bullet from Julia. The [build script for the Julia package builds the C++ API.](https://github.com/SimonDanisch/Bullet3Builder/blob/master/build_tarballs.jl#L46) I would like it to build the C API instead (or in addition) since it is easier to wrap the C API right now.

goretkin, 2019-06-01:
Closed by https://github.com/bulletphysics/bullet3/pull/2221

closed 2019-06-01 by goretkin