bulletphysics/bullet3 #2223: C++ collision detection: Use Bullet3Collision, etc. or the Bullet 2.x API?

wbthomason, 2019-04-29:
For a new C++ project using Bullet **only** for collision detection (i.e. no dynamics), is it better to use structures from the `src/Bullet3Collision`, `src/Bullet3Common`, etc. directories, or to use the "old" structures (e.g. `btVector3` instead of `b3Vector3`, `btDbvtBroadphase` instead of `b3DynamicBvhBroadphase`, etc.)?

hermestrimegistus, 2019-05-15:
I really don't know the difference myself. What is btVector3 and what is b3Vector3? Which is meant for programmer's use and which is there for api designer?

phcerdan, 2019-11-05:
I would like to clarify this too. @wbthomason did you figure it out?

wbthomason, 2019-11-05:
@phcerdan unfortunately not; I had to get my project using Bullet finished quickly, so I didn't spend much time after this issue looking into the `Bullet3` options. I just used the structures from the old C++ documentation.

I too would still be interested in clarifying this for using Bullet moving forward.

Satya758, 2020-05-17:
https://github.com/bulletphysics/bullet3/issues/1401 has information.

closed 2021-04-26 by erwincoumans