bulletphysics/bullet3 #2228: CCD doesn't work

AndreaCatania, 2019-05-02:
I've integrated the rigidbody CCD in godot engine and the function that calculate the threshold is this one:

```
void RigidBodyBullet::set_continuous_collision_detection(bool p_enable) {
	if (p_enable) {

		btBody->setCcdMotionThreshold(1e-7);

		/// Calculate using the rule writte below the CCD swept sphere radius
		///     CCD works on an embedded sphere of radius, make sure this radius
		///     is embedded inside the convex objects, preferably smaller:
		///     for an object of dimensions 1 meter, try 0.2
		btScalar radius(1.0);
		if (btBody->getCollisionShape()) {
			btVector3 center;
			btBody->getCollisionShape()->getBoundingSphere(center, radius);
		}
		btBody->setCcdSweptSphereRadius(radius * 0.2);
	} else {
		btBody->setCcdMotionThreshold(10000.0);
		btBody->setCcdSweptSphereRadius(0.);
	}
}
```

The scene has a rigibody with a capsule has shape, and in front of it there's a wall with a box shape.
The distance from each other is about 10 unit, and the physics FPS are 60.
When I press space I apply an inpulse of 500 and even with ccd enabled it pass the wall.

Now the question is if I'm doing something wrong to calculate the thresholds, or if I have to do something more to activate the CCD test.

Here the related issue in Godot: https://github.com/godotengine/godot/issues/16113

[image]

Steven89Liu, 2019-05-16:
i write a simple test with similar parameters as yours, but i can't reproduce this issue. Could you help to check if there is some difference between us.
1. create the rigidbody
   `testBody = createRigidBody(mass,startTransform,colShape);
    testBody->setCcdMotionThreshold(1e-7);
    testBody->setCcdSweptSphereRadius(radius*0.2);`
2. change the velocity by the key
`bool BasicExample::keyboardCallback(int key, int state) {
        //key 1, key 2, key 3
        printf("key = %d, state=%d\n", key, state);
        if (state == 0) return false;
        testBody->setActivationState(1);
        switch (key) {
        case '1' /*ASCII for 1*/: {
                printf("key 1\n");
                testBody->setLinearVelocity(btVector3(200, 0, 0));
                return true;
        }
        case '2' /*ASCII for 2*/: {
                printf("key 2\n");
                testBody->setLinearVelocity(btVector3(500, 0, 0));
                return true;
        }
        case '3' /*ASCII for 3*/: {
                printf("key 3\n");
                testBody->setLinearVelocity(btVector3(10000, 0, 0));
                return true;
        }
        }

        return false;
}
`
[image]

the test file is attached. 
[BasicExample.cpp.txt](https://github.com/bulletphysics/bullet3/files/3185634/BasicExample.cpp.txt)

AndreaCatania, 2019-05-16:
Thanks for the answer I'll do a merge of bullet physics inside Godot soon, after this we can start to see why it doesn't work so we can work on a synchronized version of Bullet. I let you know.

gururise, 2019-11-26:
Here's a simple test project demonstrating the issue:
[image]

Make sure you have Debug->Visible collision shapes enabled

[rb3dtest.zip](https://github.com/bulletphysics/bullet3/files/3893592/rb3dtest.zip)

closed 2021-04-26 by erwincoumans