bulletphysics/bullet3 #2229: link error with -DBUILD_OPENGL3_DEMOS=OFF

goretkin, 2019-05-03:
On master, on macOS and Ubuntu I get the following link errors:

```
[ 18%] Built target OpenGLWindow
[ 18%] Linking CXX executable AppBasicExampleGui
Undefined symbols for architecture x86_64:
  "SimpleOpenGL3App::SimpleOpenGL3App(char const*, int, int, bool, int, int, int, int)", referenced from:
      _main in main_opengl_single_example.o
  "_useShadowMap", referenced from:
      GL_ShapeDrawer::drawScene(btDiscreteDynamicsWorld const*, bool, int) in GL_ShapeDrawer.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [examples/BasicDemo/AppBasicExampleGui] Error 1
make[1]: *** [examples/BasicDemo/CMakeFiles/AppBasicExampleGui.dir/all] Error 2
make: *** [all] Error 2
```

when building with the following cmake command:

```
cmake -DBUILD_PYBULLET=ON -DBUILD_PYBULLET_NUMPY=ON -DUSE_DOUBLE_PRECISION=ON -DBT_USE_EGL=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_OPENGL3_DEMOS=OFF ..
```

The only difference from https://github.com/bulletphysics/bullet3/blob/de2d22f76bcb9f8339a00233f51525ef654d6714/build_cmake_pybullet_double.sh#L8 is the `-DBUILD_OPENGL3_DEMOS=OFF` flag.

I can't really explain why this happens, because the only branch I see on that flag is: https://github.com/bulletphysics/bullet3/blob/233a7083d19ded151e80c2d9c7cd3d28856d6f76/CMakeLists.txt#L397

erwincoumans, 2019-05-03:
Then leave it on. We should remove that flag.

goretkin, 2019-05-03:
I'm not sure if this was the intention of the flag originally, but I would like a way to build just pybullet, or just the c api (https://github.com/bulletphysics/bullet3/pull/2221).

erwincoumans, 2019-05-03:
pip install pybullet does that. Or use premake/cmake and only build the pybullet target.

closed 2019-05-06 by goretkin

erwincoumans, 2019-05-06:
Also, PyBullet is mainly implemented as a collection of 'examples' with some glue code in pybullet.c and examples/SharedMemory. For example, the URDF importer can be used as a separate example or as part of PyBullet.