bulletphysics/bullet3 #2237: Velocity control for Robot base via p.setJointMotorContro2Array()

arpastrana, 2019-05-06:
## Description
The goal is to control a "robotless" floating object in space with **velocities** as opposed to **positions**.

## Expected Result
Using `p.setJointMotorContro2Array()` in `p.VELOCITY_CONTROL` mode, the robotless object moves accurately in the world coordinate system.

[image]

## Actual Result: Changing Constraint
As it had been discussed in the forum (https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=12114) and shown in pybullet's `examples/vrhand.py`, we have first tried to set a **constraint** at the base of the loaded `.urdf`file, then calculate position and orientation deltas by multiplying a user sent velocity times the`time_step`, added them to the current base position/orientation, to ultimately pass these values to `p.changeConstraint()`.  

A code snippet of this process looks as follows:
```
def apply_action_change_constraint(velocity):
        pos_lin = np.array(velocity[3:6]) * time_step
        pos_rot = np.array(velocity[3:6]) * time_step

        base_pos, base_orn = p.getBasePositionAndOrientation(robotless)
        new_pos = np.add(np.array(base_pos), pos_lin).tolist()
        new_orn = np.add(np.array(p.getEulerFromQuaternion(base_orn)), pos_rot)
        new_orn = p.getQuaternionFromEuler(new_orn.tolist())

        p.changeConstraint(robotless.base_constraint, new_pos, new_orn)
```

This method did help us achieve a coherent way to displace our robotless object in space. However, over different experiments, we noticed a downscaling in the velocities when comparing the sent velocity to the velocity at the base (`velocity` / `p.getBaseVelocity()` = `5`) . 

## Actual Result: Motor Control
In a next step, as suggested by @erwincoumans, we have attempted to control it via `p.setJointMotorContro2Array()`. Our implementation has introduced virtual links (mass=0.001, inertia=0.001), and one joint per desired velocity component in the `.urdf` file (linear_x, linear_y, linear_z, angular_x, angular_y, angular_z). The joints are of type `prismatic` for the linear components, and of type `continuous` for the angular ones. The object's base has been fixed at the world frame. 

Our implementation goes along the lines of:

```
def apply_action_motor_control(velocity):
        p.setJointMotorControlArray(bodyUniqueId=robotless.id, 
                                    jointIndices=list(range(6)),
                                    controlMode=p.VELOCITY_CONTROL,
                                    targetVelocities=velocity
                                    )
```

We noticed that the object will comply to the orientation of the frame at which the base is initially loaded at, **except** for the last joint defined in the `.urdf` file. This last one will move on the _displaced_ robotless local frame. We have tried reducing the number of joints, and changing the joint type (from prismatic to continuous and viceversa), but the issue persists.

The problem is evident, for instance, when after rotating first **90 degrees** over the base **Y-axis**, we sent an angular velocity for the object to rotate next around the **Z-axis**. One would expect it to move around **global Z** (as the base frame and world frame coincide), but in fact it does revolve around the new position of **local Z** (which now matches **global X**).

## Files
The files for our reproduction case are here: [motor_control_test.zip](https://github.com/bulletphysics/bullet3/files/3149592/motor_control_test.zip)

## Questions
- How could we go about the `p.setJointMotorControlArray()` issue we are facing so that we get frame-consistent velocity control over our object?

Thanks in advance.

Steven89Liu, 2019-05-15:
"except for the last joint defined in the .urdf file. "
All the translation and rotation are relative to its parent coordinate system.  all your parameters are relative value, but seems that what you want is absolute value relative to the fix frame, i think you should transform these absolute values to relative values for each joint.

closed 2019-07-10 by arpastrana

sahandrez, 2020-07-06:
@arpastrana How did you solve this issue?

arpastrana, 2020-07-06:
@sahandrez We didn’t solve it. For our experiments, we ended up using `p.changeConstraint()`, making it clear in the docs that `velocity / p.getBaseVelocity() = 5`.

sahandrez, 2020-07-06:
@arpastrana Ok thanks!