bulletphysics/bullet3 #2240: Depth Calculation issue in DIRECT mode; Increasing simulation speed for RL

adityajain07, 2019-05-07:
[binpacking.tar.gz](https://github.com/bulletphysics/bullet3/files/3152036/binpacking.tar.gz)

[image]

Hi!

I am working on a simulation wherein I have to add various sizes of boxes in a container, one-by-one (please see the attached snapshot). Some points:

- The container (green colour) is imported through an SDF file (please see zip folder)
- The boxes are created using the PyBullet createMultiBody function which fetches the box dimensions from 'box-list.txt' file
- There is a camera on top (along z-axis i.e. bird's eye view) which captures this view of the container and calculates the depth map

### First Problem: Depth Issue in DIRECT mode
If I run the 'BinPacking-GUI.py' file, the simulation runs in the GUI mode and it gives me the correct depth map. 
Note: I am printing the maximum height from the base i.e. (cameraHeight - depthMap)
For the testing purpose, I was adding boxes of larger height of the container and the code gives correct value.

However, when I run the same in DIRECT mode using 'BinPacking-DIRECT.py' file, it only takes into account the container (which was imported using the SDF file) and completely ignores the boxes being added.

### Second Issue: Speeding up simulation
My project involves running a Deep-RL algorithm on this simulation. I only want to add the next box when my current system is stable (there can be cases when one box topples over the other). Thus in line 184, I am running stepSimulation() function for few loops (arbitarily decided). I have also set p.setPhysicsEngineParameter(fixedTimeStep=1/60, numSolverIterations=10), to increase the simulation speed. 
However, still it takes approx. 1 sec to add each box in the DIRECT mode. Thus running this simulation for millions of episodes and also adding the time required to train the network, will be huge.

Please let me know, if I am missing something obvious.

Thanks,
Aditya

Steven89Liu, 2019-05-15:
i find that the engine is not so good at supporting the GEOM_PLANE in DIRECT mode. so the initial state you get is not right.  you can try to replace the plane with the box like below. Good luck!
        planeId = p.createCollisionShape(p.GEOM_BOX,halfExtents=[100,100,0.001])
        p.createMultiBody(0, planeId,-1, [0, 0, 0], [0, 0, 0, 1])

erwincoumans, 2019-05-31:
For fastest reset speed, avoid loading/creating new bodies, but reposition them (pre-create many bodies if different sizes). Keep the unused somewhere far away.

It could be a tiny renderer issue, not supporting programmatic creation. I'll look into your file when I find some time.

adityajain07, 2019-06-03:
@erwincoumans, thank you for your response. Regarding the depth issue in DIRECT mode, when I raised the camera height to 100 units from the base (from the earlier 5), it apparently gave the correct values. Not sure why it fixed that. Does it have to do with the 'near' and 'far' values we set?

And thanks for the comment on the speed-up issue. Will give it a shot :)

closed 2021-04-26 by erwincoumans