bulletphysics/bullet3 #2245: btMultiBody constraint limits

AndreaCatania, 2019-05-08:
I'm implementing the multi body inside Godot Engine, and the only problem that I'm facing is to understand how to apply the joint limits.

[image]

I've a chain of elements as you can see, connected by a revolute joints.
I would apply a lower and upper limits of 5 degree and to do it I've tried many approaches foe example:

```
getLink(get_link_id()).m_jointLowerLimit = -q0;
getLink(get_link_id()).m_jointLowerLimit = q0;
```

or 
`setJointPosMultiDof(get_link_id(), &q0);`

I have used 
`addLinkConstraintForce`

And non of them seems to work.

I'm starting to thing (even if it seems a bit strange) that I've to implement the limits myself by setting the forces manually.

I would really appreciate if someone could point out the right way to implement them.

Thanks Andrea

AndreaCatania, 2019-05-08:
Another test that I did was to reset the position each step using:

```

	btScalar q0 = 50.f * SIMD_PI / 180.f;
	setJointPosMultiDof(get_link_id(), &q0);
```

and the result is that the contact responses are not correctly propagated.
[image]

AndreaCatania, 2019-05-08:
Just figured out that the joint constraint must be set using this: btMultiBodyJointLimitConstraint

Thanks for the Help!

closed 2019-05-08 by AndreaCatania

goodnamegood, 2020-04-15:
@AndreaCatania did you find a way to do this in pybullet? btMultiBodyJointLimitConstraint is not a pybullet function.

erwincoumans, 2020-04-15:
@goodnamegood  PyBullet will load URDF files and apply the joint limits using btMultiBodyJointLimitConstraint  constraints, as specified in the URDF file. Make sure to use "revolute" joint type and not "continuous".