bulletphysics/bullet3 #2246: Numerical difference between pybullet and RBDL/KDL

lillmaria, 2019-05-08:
Hi! 

I am using pybullet, RBDL and KDL to evaluate the inverse dynamics and inertia matrix of a kuka-lwr and a ur5. When comparing 1000 evaluations of the inverse dynamics/inertia matrix, the numerical difference between RBDL and KDL is 10^(-12), while the difference between KDL/RBDL and pybullet is 10^(-3). 

I use a standard kuka-lwr urdf and ur5 urdf to load the robot model for all the libraries. I have also enabled the URDF_USE_INERTIA_FROM_FILE flag. 

Also, I have noticed that if all the joint origins and link origins are changed to (0, 0, 0), or close to this, the numerical difference decreases to 10^(-12).  

Du you know why the numerical difference is so "big" for the original urdfs?

erwincoumans, 2019-05-08:
Without access to your test it is impossible to tell where the differences are. Is your test available somewhere?

lillmaria, 2019-05-09:
After some testing, it seems to me that the problem is related to integers vs. decimals. As long as the urdf's origins consist of integers the numerical difference is of factor 10^(-12), while when changing just one element from integer to a decimal, the numerical difference is of factor 10^(-3), sometimes 10^(-2). 

To demonstrate a minimal example, I use a pantilt-urdf and a test script to compare gravitational impact on the joints using the inverse dynamics functionality: 

```

import rbdl
import numpy as np
import PyKDL as kdl
import kdl_parser_py.urdf as kdlurdf
import pybullet as pb

root = 'base_link'
tip = 'tilt_link'
path_to_urdf = 'pantilt.urdf'
ok, tree = kdlurdf.treeFromFile(path_to_urdf)

kdlmodel = tree.getChain(root,tip)
rbdlmodel = rbdl.loadModel(path_to_urdf)

sim = pb.connect(pb.DIRECT)
pbmodel = pb.loadURDF(path_to_urdf, useFixedBase=True, flags = pb.URDF_USE_INERTIA_FROM_FILE)
pb.setGravity(0, 0, -9.81)
q_max = [3.14, 4.64]
q_min = [-3.13, -4.64]
n_joints = 2

q_kdl = kdl.JntArray(n_joints)
tau_kdl = kdl.JntArray(n_joints)
gravity_kdl = kdl.Vector()
gravity_kdl[2] = -9.81

q_pb = [None]*n_joints
zeros_pb = [None]*n_joints

q_rbdl = np.zeros(n_joints)
tau_rbdl = np.zeros(n_joints)
zeros_rbdl = np.zeros(n_joints)

error_kdl_rbdl = np.zeros(n_joints)
error_pb_rbdl = np.zeros(n_joints)
error_pb_kdl = np.zeros(n_joints)

def list2np(asd):
    return np.asarray(asd)

n_itr = 1000

for i in range(n_itr):
    for j in range(n_joints):
        q_pb[j] = (q_max[j] - q_min[j])*np.random.rand()-(q_max[j] - q_min[j])/2
        q_kdl[j] = q_pb[j]
        q_rbdl[j] = q_pb[j]
        zeros_pb[j] = 0.

    rbdl.InverseDynamics(rbdlmodel, q_rbdl, zeros_rbdl, zeros_rbdl, tau_rbdl)
    kdl.ChainDynParam(kdlmodel, gravity_kdl).JntToGravity(q_kdl, tau_kdl)
    tau_pb = pb.calculateInverseDynamics(pbmodel, q_pb, zeros_pb, zeros_pb)

    for tau_idx in range(n_joints):
        error_kdl_rbdl[tau_idx] += np.absolute((list2np(tau_kdl[tau_idx]) - tau_rbdl[tau_idx]))
        error_pb_rbdl[tau_idx] += np.absolute((list2np(tau_pb[tau_idx]) - (tau_rbdl[tau_idx])))
        error_pb_kdl[tau_idx] += np.absolute((list2np(tau_kdl[tau_idx]) - list2np(tau_pb[tau_idx])))

sum_error_kdl_rbdl = 0
sum_error_pb_rbdl = 0
sum_error_pb_kdl = 0

for err in range(n_joints):
    sum_error_kdl_rbdl += error_kdl_rbdl[err]
    sum_error_pb_kdl += error_pb_kdl[err]
    sum_error_pb_rbdl += error_pb_rbdl[err]

print "\nSum of errors KDL vs. RBDL for", n_itr, "iterations:\n", sum_error_kdl_rbdl
print "\nSum of errors KDL vs. pybullet for", n_itr, "iterations:\n", sum_error_pb_kdl
print "\nSum of errors RBDL vs. pybullet for", n_itr, "iterations:\n",sum_error_pb_rbdl
```

The pantilt-urdf, taken from ROS tutorials, looks like this:

```

<?xml version="1.0"?>
<robot name="pan_tilt">

  <link name="base_link">
       <inertial>
  	<mass value="1"/>
    <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
  	<inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
  </inertial>
    </link>

    <joint name="pan_joint" type="revolute">
      <parent link="base_link"/>
      <child link="pan_link"/>
      <origin xyz="0 0 0.1"/>
      <axis xyz="0 0 1" />
      <limit effort="300" velocity="0.1" lower="-3.14" upper="3.14"/>
      <dynamics damping="50" friction="1"/>
    </joint>

    <link name="pan_link">
     <inertial>
  	<mass value="1"/>
    <origin rpy="0 0 0" xyz="0 0 1"/>
  	<inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
  </inertial>
    </link>

    <joint name="tilt_joint" type="revolute">
      <parent link="pan_link"/>
      <child link="tilt_link"/>
      <origin rpy="0.0 0.0 0.0" xyz="4 0 1"/>
      <axis xyz="0.0 1.0 0.0" />
      <limit effort="300" velocity="1.0" lower="-4.64" upper="4.64"/>
    </joint>

    <link name="tilt_link">
    <inertial>
  	<mass value="5"/>
    <origin rpy="0 0 0" xyz="1.001 1 1"/>
  	<inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
  </inertial>
    </link>
</robot>
```

As long as the tilt_link's xyz-origin consist of integers (doesn't matter which integers), the difference is 10^(-12) for 1000 iterations. If, for instance, the tilt_link's origin-xyz changes from (1 1 1) to (1.001 1 1), the numerical difference becomes 10^(-2). Do you know why this happens? The decimals in the origins are also the reason the numerical difference for the kuka-lwr and the ur5 becomes so big. Is it something with pybullet's datatypes?

erwincoumans, 2019-05-09:
Thanks for the repro case, I can see the difference between rbdl and pybullet.
It also reminds me how much headache it is to install KDP and rbdl, for various Python versions on Mac OSX. Why can't they just have pip install PyKDL and pip install rbdl working out-of-the-box like pybullet?

I'll have a look when I find time, to check it out. It may be the difference between single precision float and double precision float (some parts of the pybullet URDF importer may use use single-precision)

erwincoumans, 2019-05-09:
A sum of error of 10^(-3) over 1000 iterations, is in the order of 10^-6 per step. It is not clear yet where the difference happens. Bullet uses double precision. Does rbdl/PyKDL also use double precision floats?

lillmaria, 2019-05-14:
PyKDL and rbdl use double precision, so that is probably not it then. What are the other possible options for this difference?

erwincoumans, 2019-05-14:
The difference per step is small (1e-6) so it seems like somewhere a double to single precision conversion happens. It could also be some floating point cancellation issue (adding a small number to a larger number gets lost).

It is too small of a difference to track down urgently for me. Feel free to dig into it yourself, it is all open source.

Steven89Liu, 2019-06-19:
there is conversion from the double to single in the UrdfParser.cpp, i will submit this change later.

erwincoumans, 2019-06-19:
change should be submitted, let's close the issue (re-open if not fixed after next pybullet update)

closed 2019-06-19 by erwincoumans