bulletphysics/bullet3 #2250: Kuka robot arm pybullet env doesn't render?

maximecb, 2019-05-13:
Hi. I am not seeing any visual output for the kuka env, when calling `env.render('human')`.

Test code to reproduce:

```
import gym
import pybullet_envs
import numpy as np
import time
env = gym.make('KukaCamBulletEnv-v0')
env.reset()
env.render('human')

for i in range(50):
    action = np.random.uniform(low=-1, high=1, size=(env.action_space.shape[0],))
    obs, reward, done, info = env.step(action)
    time.sleep(0.2)
    env.render('human')
```

The same code works for the reacher environment, with only the environment name changed.

maximecb, 2019-05-13:
Upon further examination, the reacher env does produce visual output, but the camera is very far. There is a method `camera_adjust` [defined here](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/gym_manipulator_envs.py#L7), but I think this method is never called, because when I call it, there seems to be a bug in the method:

```
  File "<stdin>", line 1, in <module>
  File "/home/maximecb/.local/lib/python3.6/site-packages/pybullet_envs/gym_manipulator_envs.py", line 44, in camera_adjust
    self.camera.move_and_look_at(0.3, 0.3, 0.3, x, y, z)
  File "/home/maximecb/.local/lib/python3.6/site-packages/pybullet_envs/env_bases.py", line 138, in move_and_look_at
    self._p.resetDebugVisualizerCamera(distance, yaw, -20, lookat)
AttributeError: 'Camera' object has no attribute '_p'
```

maximecb, 2019-05-13:
I am able to correct the camera position in the reacher environment manually like this:
```
import pybullet
pybullet.resetDebugVisualizerCamera(0.5, 0, -45, (0, 0, 0))
```

erwincoumans, 2019-05-13:
Can you call env. render('human') once before the first env. reset? The issue is that PyBullet needs to create an OpenGL window at initialization. Will be fixed in a future major revision.

maximecb, 2019-05-13:
@erwincoumans I still get no window showing on the kuka env, whether I call env.render() before or after reset.

maximecb, 2019-05-24:
Ping @erwincoumans. Can you check that you can get the kuka env to show a render window?

Teslatic, 2019-11-05:
Hey @maximecb,
I stumbled on this issue as well and found a workaround. Maybe it comes a little bit late, but maybe it can help others:

`
from pybullet_envs.bullet.kukaGymEnv import KukaGymEnv

from itertools import count

env = KukaGymEnv(renders=True, isDiscrete=False)
env.reset()

for t in count():
    action = env.action_space.sample()

    next_obs, reward, done, _ = env.step(action)

    if done == True:
        break

    obs = next_obs
`

swagatk, 2020-10-09:
Dear @maximecb, thanks for your code. It does work and I am able to see the simulation environment. Does this mean that I can now implement my RL algorithm on this?  Where can I find more information about various action and observation fields used by this simulation? Is there any documentation available for this environment? I am using the Cam version of this environment: `KukaCamGymEnv`.

[image]

closed 2021-04-26 by erwincoumans