bulletphysics/bullet3 #2252: Access `btDiscreteDynamicsWorld` from a `PhysicsClient`

goretkin, 2019-05-14:
I am trying to use Bullet from the Julia language, which has great interoperability with C and some decent interoperability with C++. I would like to try to use both of these APIs simultaneously.

With the Bullet C API, I can get a `b3PhysicsClientHandle`, which I can then cast to a `PhysicsClient`, as is done in several places: https://github.com/bulletphysics/bullet3/blob/7b06293a97eba4a552a698e822f072fec717f30b/examples/SharedMemory/PhysicsClientC_API.cpp#L14

I would now like to get access to the `btDiscreteDynamicsWorld` object that underlies the interface, so that I can use the C++ API, which is in `m_data->m_dynamicsWorld`.

Is there an example that uses both APIs simultaneously? If not, does it seem like the it is possible to do so currently? Note that I am only interested in doing this with the `PhysicsDirect` class, so that everything is in the same process address space.

In `PhysicsDirect`, the `m_data` field is `protected`, so I cannot access it.

goretkin, 2019-05-14:
If I'm understanding the gist of https://github.com/bulletphysics/bullet3/issues/1328 correctly, the C++ API is not really intended to be used anymore. It would still be great to have access to it for debugging purposes, however.

goretkin, 2019-05-15:
I was able to use the private member in my case. For anyone else giving it a try, the relevant chain, I believe, is

`((PhysicsServerCommandProcessor*) ((PhysicsDirect*) sm)->m_data->m_commandProcessor)->m_data->m_dynamicsWorld`

where `sm` is the `b3PhysicsClientHandle`.

Here is an example of accessing the C and C++ API from Julia: 
https://github.com/goretkin/Bullet.jl/blob/d85f3cb8ea005c1bcc87626b09d37389c742bca1/test/cxx_bullet_c_api.jl

closed 2019-05-15 by goretkin

erwincoumans, 2019-05-15:
Yes, best to keep it private, the intention is to have a higher level C API with Python bindings that allows switching physics engines, add IK, inverse dynamics, VR and rendering for experiments.