bulletphysics/bullet3 #2254: Not able to install

AYUSHKABIRVERMA, 2019-05-16:


AYUSHKABIRVERMA, 2019-05-23:
Installing via conda solved my issue
conda install -c conda-forge bullet

closed 2019-05-23 by AYUSHKABIRVERMA

erwincoumans, 2019-05-23:
bullet 2.88 is not the same as pybullet. You can build PyBullet using visual studio manually, see the PyBullet quickstart guide.

AdarshPan, 2020-11-05:
I am also not able to install pybullet . Getting a lot of error .

    copying examples\pybullet\gym\pybullet_data\random_urdfs\986\986.urdf -> bui
ld\lib.win-amd64-3.7\pybullet_data\random_urdfs\986
   
    creating build\lib.win-amd64-3.7\pybullet_data\random_urdfs\992
   
d\lib.win-amd64-3.7\pybullet_data\random_urdfs\993
    copying examples\pybullet\gym\pybullet_data\random_urdfs\993\993.urdf -> bui
ld\lib.win-amd64-3.7\pybullet_data\random_urdfs\993
    creating build\lib.win-amd64-3.7\pybullet_data\random_urdfs\994
    copying examples\pybullet\gym\pybullet_data\random_urdfs\994\994.mtl -> buil
d\lib.win-amd64-3.7\pybullet_data\random_urdfs\994
   
m6\collision\link6_vhacd.obj -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_
description\meshes\xarm6\collision
    creating build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_description\meshes\
xarm6\visual
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_description\meshes\xar
m6\vi
on\meshes\xarm6\visual
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_description\meshes\xar
m6\visual\link6.stl -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_descripti
on\meshes\xarm6\visual
    creating build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper
    creating build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\meshes
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_gripper\meshes\base_li
nk.STL -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\meshes
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_gripper\meshes\left_fi
nger.STL -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\meshes
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_gripper\meshes\left_in
ner_knuckle.STL -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\meshe
s
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_gripper\meshes\left_ou
ter_knuckle.STL -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\meshe
s
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_gripper\meshes\right_f
inger.STL -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\meshes
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_gripper\meshes\right_i
nner_knuckle.STL -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\mesh
es
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_gripper\meshes\right_o
uter_knuckle.STL -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\mesh
es
    running build_ext
    building 'pybullet' extension
    error: [WinError 3] The system cannot find the path specified: 'C:\\Program
Files (x86)\\Microsoft Visual Studio 14.0\\VC\\PlatformSDK\\lib'
    ----------------------------------------
ERROR: Command "'C:\Users\SHRIKANT\Anaconda3\python.exe' -u -c 'import setuptool
s, tokenize;__file__='"'"'C:\\Users\\SHRIKANT\\AppData\\Local\\Temp\\pip-install
-hbfo57c2\\pybullet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__f
ile__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compil
e(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\SHRIKANT\AppData\
Local\Temp\pip-record-omd21xdj\install-record.txt' --single-version-externally-m
anaged --compile" failed with error code 1 in C:\Users\SHRIKANT\AppData\Local\Te
mp\pip-install-hbfo57c2\pybullet\

erwincoumans, 2020-11-05:
Just reinstall visual studio and make sure to use a MSVC command prompt, not a regular terminal.

AdarshPan, 2020-11-06:
but visual studio is 7.13 GB . Not enough space  in my PC

AdarshPan, 2020-11-07:
what should i do now ??

erwincoumans, 2020-11-07:
Start saving for a larger harddisk.