bulletphysics/bullet3 #2256: btGetNumHardwareThreads can return 0

mtheall, 2019-05-16:
The C++11 version does
`return btMin<int>(BT_MAX_THREAD_COUNT, std::thread::hardware_concurrency());`
but `std::thread::hardware_concurrency()` is allowed to return 0 (and does in my specific case).

I would normally be okay with this but `btThreadSupportPosix::startThreads` uses `btGetNumHardwareThreads() - 1` for the number of threads, becoming -1 stored into an int which is also okay. What's not okay is when it does `m_activeThreadStatus.resize(m_numThreads);` which converts to unsigned aka a really big number and my app freezes.

erwincoumans, 2019-05-16:
Good point. btGetNumHardwareThreads should at least return 1, since without a CPU/thread we cannot do much. Do you care for a pull request with a fix?

erwincoumans, 2019-05-27:
I merged your PR, thanks!

closed 2019-05-27 by erwincoumans