bulletphysics/bullet3 #2259: Overwriting a variable for no reason

Timofffee, 2019-05-17:
Good day. I would like to know why the variable in this place is overwritten? It looks very strange.

File/Line: https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/ConstraintSolver/btContactConstraint.cpp#L141

Steven89Liu, 2019-05-17:
they should get the same value, all implement the equation: rel_vel = (v0 + w0 x r0 - v1 - w1 x r1).dot(n). 
it seems that the btJacobianEntry is obsolete function not used at all.

Timofffee, 2019-05-17:
@Steven89Liu I was interested in the function btJacobianEntry and why its result is overwritten in such a "barbaric way".

Steven89Liu, 2019-05-17:
in my view, you can delete these code if you like.

Timofffee, 2019-05-17:
Ahaha. Optimistically. Thanks.

closed 2019-05-17 by Timofffee