bulletphysics/bullet3 #2262: SDF ball joints

C-monC, 2019-05-21:
Good day,

I'm trying out all the joints but cannot get the ball joint to work in SDF using `pybullet.loadSDF("model.sdf")`. The following code works if I change type to "revolute", it defaults to axis <1 0 0>.

`<joint name="Ball1" type="ball">`
			`    <parent>RigidGroup3</parent>`
			`    <child>RigidGroup4</child>`
			`    <pose>0 0 0 0 0 0</pose>`
`</joint>`

Should I stack revolute joints as a workaround?

closed 2019-05-21 by C-monC

erwincoumans, 2019-05-21:
PyBullet doesn't support 'ball' joint indeed in SDF. We do have 'spherical' in URDF though, see 
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/humanoid/humanoid.urdf

Try it out:
pip install pybullet
python3 -m pybullet_envs.deep_mimic.testrl --arg_file run_humanoid3d_backflip_args.txt

C-monC, 2019-05-22:
Ah okay thanks for the help.

ChristianBecker90, 2020-09-11:
I just read this questions and i also do have some kind of problem with my .sdf file. 
A little bit of my backround, i want to simulate and train a ball-on-plate model which i created via CAD (Fusion360). 
The model has three ball joints on the bottomside of the plate, one in the middle of the plate, one at the right edge of the plate and one at the back edge of the plate. Under both ball joints at the edges is one servomotor to controll the angle of the plate in one direction. 

The export via SDFusion works more or less pretty well. I think the problem are the kinematic chains, which are owed to the structure of the ball-on-plate-model.

ChristianBecker90, 2020-09-11:
...wasn´t finished :D

So i did a little research to solve my problem, but i just found some infos about the difference between SDF (kinematic loops are possible) and URDF (kinematic loops are not possible, but with an additional .py file it should run somehow). Know can anybody tell me something about how to solve my challenge?

Thanks for your support,
best regards chris

C-monC, 2020-09-16:
@ChristianBecker90 Using constraints in Pybullet it's possible to create closed loop URDF robots. Check this [project](https://github.com/Einsbon/delta-robot-pybullet/tree/master/tripedal-delta-robot) if you want to see an implementation.

rohit-kumar-j, 2020-12-27:
> I just read this questions and i also do have some kind of problem with my .sdf file.
> A little bit of my backround, i want to simulate and train a ball-on-plate model which i created via CAD [...]

Hi! @ChristianBecker90 Could you link the .sdf file converter for fusion 360? Is it an Add-in like [this](https://github.com/syuntoku14/fusion2urdf)?