bulletphysics/bullet3 #2269: SIGSEGV in btSparseSdf<3>::Evaluate()

stephengold, 2019-05-27:
In 64-bit Linux, I created a `btSoftRigidDynamicsWorld` and added one `btRigidBody` and one `btSoftBody` with `m_kAST=0`. I joined them with 2 anchors, each with `influence=4`. In the btSoftBody::Config struct, I set `piterations=3`.

During simulation, the soft body tore itself to shreds, which is fine. The issue is that the application crashed in `btSparseSdf`. The crash is reproducible. The crash does not occur when the anchors have `influence=1`. I built using a clone of the bullet3 repo at SHA1 ID=26486d56. I will try older versions to determine whether this is a recent regression.

Here is a partial stack trace:
```
  btSparseSdf<3>::Evaluate(btVector3 const&, btCollisionShape const*, btVector3&, float)+0x3d9
  btSoftBody::checkContact(btCollisionObjectWrapper const*, btVector3 const&, float, btSoftBody::sCti&) const+0xc7
  btSoftColliders::CollideSDF_RS::DoNode(btSoftBody::Node&) const+0xc8
  btSoftColliders::CollideSDF_RS::Process(btDbvtNode const*)+0x2f
  btDbvt::collideTV(btDbvtNode const*, btDbvtAabbMm const&, btDbvt::ICollide&) const+0x1d6
  btSoftBody::defaultCollisionHandler(btCollisionObjectWrapper const*)+0x2d0
  btDefaultSoftBodySolver::processCollision(btSoftBody*, btCollisionObjectWrapper const*)+0x27
  btSoftRigidCollisionAlgorithm::processCollision(btCollisionObjectWrapper const*, btCollisionObjectWrapper const*, btDispatcherInfo const&, btManifoldResult*)+0xda
  btCollisionDispatcher::defaultNearCallback(btBroadphasePair&, btCollisionDispatcher&, btDispatcherInfo const&)+0x1fc
  btCollisionPairCallback::processOverlap(btBroadphasePair&)+0x40
  btHashedOverlappingPairCache::processAllOverlappingPairs(btOverlapCallback*, btDispatcher*)+0x8f
  btHashedOverlappingPairCache::processAllOverlappingPairs(btOverlapCallback*, btDispatcher*, btDispatcherInfo const&)+0x2b0
  btCollisionDispatcher::dispatchAllCollisionPairs(btOverlappingPairCache*, btDispatcherInfo const&, btDispatcher*)+0x72
  btCollisionWorld::performDiscreteCollisionDetection()+0xe1
  btDiscreteDynamicsWorld::internalSingleStepSimulation(float)+0x10c
  btSoftRigidDynamicsWorld::internalSingleStepSimulation(float)+0x89
  btDiscreteDynamicsWorld::stepSimulation(float, int, float)+0x22a
  jmePhysicsSpace::stepSimulation(float, int, float)+0x4e
```

stephengold, 2019-05-27:
When built using Bullet v2.88 (clone of the bullet3 repo at SHA1 ID=126b6762) I see a very similar `SIGSEGV`. It occurs at line 229 in BulletSoftBody/btSparseSDF.h:
```
		const btScalar d[] = {c->d[o[0] + 0][o[1] + 0][o[2] + 0],
							  c->d[o[0] + 1][o[1] + 0][o[2] + 0],
							  c->d[o[0] + 1][o[1] + 1][o[2] + 0],
							  c->d[o[0] + 0][o[1] + 1][o[2] + 0],
							  c->d[o[0] + 0][o[1] + 0][o[2] + 1],
							  c->d[o[0] + 1][o[1] + 0][o[2] + 1],
							  c->d[o[0] + 1][o[1] + 1][o[2] + 1],
							  c->d[o[0] + 0][o[1] + 1][o[2] + 1]};
```
The value of `o` is `{-2147483648, -2147483648, 2}` which seems seriously wrong.

A similar crash occurs in 64-bit Windows, though not a `SIGSEGV` of course.

erwincoumans, 2019-05-27:
Thanks for the report. Can you provide a small example that is trivial to build to reproduce the issue? (using the ExampleBrowser for example)

stephengold, 2019-05-27:
Sorry, I'm working mostly in Java these days and have never used ExampleBrowser. Producing a self-contained example in C++ would be a big challenge for me.

It looks like `Evaluate()` converts floating-point numbers to array indices without any range checking. Scary!

erwincoumans, 2019-05-27:
The soft body code is an unsupported contribution, and there hasn't been much active development.
It is likely related to the specific soft body data (vertices, indices etc) you feed into Bullet.
Without repro case there is nothing we can do.

stephengold, 2019-05-27:
I'll keep that in mind. Thanks for your attention.

stephengold, 2019-05-27:
I could serialize the `btSoftMultiBodyDynamicsWorld` right before the time-step that crashes. Do you know of an application I could load the serialized data into that might reproduce the crash?

erwincoumans, 2019-05-30:
We haven't had any crashing report like this, so it could be due to data input. If you have a repro case that can be reproduced, please re-open.

stephengold, 2019-05-31:
@erwincoumans please re-open this issue for me. GitHub believes I lack permission to re-open it, probably because I'm not a member of your project.

As I indicated initially, the crash is reproducible, but it may take me a while to write a compact repro case. Since opening this issue, I've made some progress: I built and ran the example browser in my environment. I then tried modifying one of the examples to use `influence=4` in its anchors, but that wasn't sufficient to trigger a crash. I intend to pursue this approach further.

This issue is not a showstopper for me (since `influence=1` works around it in my app). However, it bothers me that `btSparseSDF.h`  lacks range checks. In addition to the risk of a crash, it seems to me data corruption could occur in some circumstances.

stephengold, 2019-06-08:
Hello? Anybody home?

Please re-open this issue.

stephengold, 2019-06-11:
Thank you. I'm still working on a compact repro case.

stephengold, 2019-11-20:
I spent a day trying to reproduce the crash in a 100% C++ testcase, but I didn't succeed. Bullet's soft-body code has moved on, and so have I, so I'm closing this issue.

closed 2019-11-20 by stephengold

erwincoumans, 2019-11-21:
Thanks for reporting / trying to reproduce it! That often takes time indeed, not very thankful work (been there, done that). Hopefully it was fixed in one of our commits.