bulletphysics/bullet3 #2271: Physx Backend With GPU Test

syslot, 2019-05-29:
Hi, I'm trying to use physx as a backend.

Using PhysX with CPU, it works well(Tested by otherPhysicsEngine.py). But using PhysX with GPU, after load urdf and do step_simulation, one exception with 'allocAligned out of sharedMemory' occurred. libPhysXGpu.so throws this exception. 

Not all of urdf would cause this exception(r2d2.urdf works normal while minitaur.urdf works abnormal). 

Is this a problem caused by URDF2PHYSX?

erwincoumans, 2019-05-29:
The PhysX backend is still early stage, and I never tried the GPU backend. You are following a new path, consider contributing :)

syslot, 2019-05-29:
Thanks for your replay. 

Based on your work and according to [the source code](https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/snippets/snippethellogrb/SnippetHelloGRB.cpp#L100) , I just do a slightly change in the PhysXServerCommandProcessor::connect. 

Now I also open [an issue in physx](https://github.com/NVIDIAGameWorks/PhysX/issues/129) and wait for some help.

erwincoumans, 2019-05-29:
This is unsupported work, best not to report issue related to this, unless you can reproduce it outside of this code base.

syslot, 2019-05-29:
Yeah . I will wait for their next version and if I test ok, I will commit one patch to the **physx4** brach 😊 .

Close it now.

closed 2019-05-29 by syslot