bulletphysics/bullet3 #2277: KukaDiverseObjectEnv's memory usage increases over time

muupan, 2019-06-02:
When using `KukaDiverseObjectEnv`, memory usage of the process seems to increase over time as I call its `reset` method, even if I call `pybullet.setPhysicsEngineParameter(enableFileCaching=0)`. I'm not sure why memory usage increases when file caching is turned off and `pybullet.resetSimulation` is called inside the `reset` method. Any help would be appreciated.

Here is the code to reproduce this issue.
```
import os
import psutil
import pybullet
from pybullet_envs.bullet.kuka_diverse_object_gym_env import KukaDiverseObjectEnv  # NOQA

env = KukaDiverseObjectEnv()
process = psutil.Process(os.getpid())
pybullet.setPhysicsEngineParameter(enableFileCaching=0)

for i in range(50):
    env.reset()
    print(i, process.memory_info().rss)
```

Output is as below, indicating memory usage increased from 133MB to 271MB.
```
0 133353472
1 146747392
2 155779072
3 163389440
4 168386560
5 172355584
6 182689792
7 186380288
8 189898752
9 194433024
10 199094272
11 200429568
12 201547776
13 205897728
14 206827520
15 207949824
16 215080960
17 220909568
18 223277056
19 226361344
20 233582592
21 235913216
22 235978752
23 236302336
24 237502464
25 242319360
26 242487296
27 243736576
28 248066048
29 249217024
30 249298944
31 252092416
32 252256256
33 254488576
34 254517248
35 261431296
36 261963776
37 266248192
38 266289152
39 266313728
40 266334208
41 266366976
42 266395648
43 270573568
44 270573568
45 270577664
46 270594048
47 270606336
48 270622720
49 271781888
```

I'm using Python 3.6.6 and pybullet==2.5.0 on macOS Mojave.

Steven89Liu, 2019-06-03:
i just try your script and don't find the issue. i'm using python 3.5 and pybullet==2.4.8 on ubuntu16.04
output is as below.
0 103555072
1 106414080
2 110960640
3 110964736
4 110968832
5 110989312
6 107376640
7 110989312
8 110964736
9 110977024
10 107360256
11 107397120
12 110968832
13 107421696
14 110960640
15 111009792
16 107352064
17 111046656

waiyc, 2019-06-07:
I am facing memory increase issue.
This is my results of running @muupan script and I am using ubuntu16.04 and pybullet==2.4.8.

>0 125063168
1 139485184
2 154886144
3 169168896
4 183762944
5 197935104
6 212467712
7 226709504
8 240656384
9 254709760
10 269189120
11 282984448
12 297275392
13 311607296
14 325709824
15 339804160
16 353624064
17 367984640
18 381902848

Steven89Liu, 2019-06-07:
sorry, the pybullet i tested with is built myself, so it's not 2.4.8.

closed 2021-04-26 by erwincoumans