bulletphysics/bullet3 #2281: Decomposing projectionMatrix into intrinsic and extrinsic matrices

Nematollahi, 2019-06-05:
Hey! I am interested in getting the intrinsic and extrinsic matrices of the simulated camera in pybullet. I assume that it should be possible by only decomposing the projectionMatrix. Could you please guide me how I can achieve this? 

I appreciate your support.

closed 2019-06-11 by Nematollahi

NotAnyMike, 2020-02-27:
Any update here?

erwincoumans, 2020-02-27:
No. You are free to bring your own view and projection matrix, as well as intrinsic and extrinsic matrices. PyBullet uses OpenGL conventions by default, but you are free to use any other perspective matrix.

NotAnyMike, 2020-02-27:
Do you know any good resource to understand how the projection matrix from openGL is calculated and  the link to the intrinsic and extrinsic?

erwincoumans, 2020-02-27:
No, I am not familiar with intrinsic and extrinsic matrices.

erwincoumans, 2020-02-27:
Google?

domef, 2023-08-17:
In the function `computeProjectionMatrixFOV` the FOV is horizontal, vertical or diagonal?

domef, 2023-08-17:
@nematoli did you solve the problem? I'm also using the RQ-factorization but the result seems wrong.