bulletphysics/bullet3 #2289: x64/VC++ 2019/btSparseSDF inefficient hash function

gjaegy, 2019-06-12:
Hi, 

following issue #2274, I've managed to find out what's causing the huge performance loss when having a kinematic rigid body colliding with a 2k nodes soft body, which appeared while switching to Visual 2019.

Actually, the key is the Hash() function at the end of the btSparseSDF.h file, especially the btS struct that's used to pack all the input into the continuous dword array that's given as input of HsiehHash.

When compiling as 64-bits, the size of the void* pointer is 8 bytes. Also, with VS2019, the pointer gets aligned, so there is a 4-byte pad between z and p.
That 4-byte pad gets filled with random values, which are taken into account in the HsiehHash() hash function when computing the hash function.
This increases the hash collisions significantly (several orders of magnitude).

A simple solution is to force the struct to not be padded 👍 
`#pragma pack(push, 1)
		struct btS
		{ 
			int x,y,z;
			void* p;
		};
#pragma pack(pop)`

I am not fully sure of the portability of the pragma pack directive however.

Another option would be to manually pack x/y/z/shape* into a raw int* buffer, but that will require special handling of the x86/x64 cases (sizeof pointer).

I can easily fix that for you. Let me know which option you prefer. Also, in the case the second one is prefered, let me know what preprocessor directive I can use to handle the different 32-64 bits cases.

gjaegy, 2019-06-13:
I've came up with a better option that gets rid of any problem mention above. Simply replace the Hash() function in btSparseSDF.h with the following one. It will work for any platform and doesn't require any manual handling of the 32/64 bits cases:

`	static inline unsigned int	Hash(int x,int y,int z,const btCollisionShape* shape)
	{
		struct btS
		{ 
			int x,y,z;
			void* p;
		};
		btS myset;

		// manually fill the 'myset' memory block in order to avoid potential padding before *p in the struct memory layout (pointer alignment) - if any padding occurs there will simply be unused bytes at the end of the struct
		int* pmyset = (int*)&myset;
		*pmyset = x; ++pmyset;
		*pmyset = y; ++pmyset;
		*pmyset = z; ++pmyset;
		*(void**)pmyset = (void*)shape;

		const void* ptr = &myset;

		unsigned int result = HsiehHash<(3*sizeof(int) + sizeof(void*))/4> (ptr);

		return result;
	}`

Erwin, shall I post a pull request or could handle this by yourself with the block of code above ?

erwincoumans, 2019-06-13:
Thanks for digging this up. If you can create a small patch that would be great. Did you test the patch with the ExampleBrowser softbody examples?

gjaegy, 2019-06-20:
Sure. I've zipped the patch in order to be able to attach it in that message. 
Please let me know if that works or not.

[btsparseSDF_imagine_patch.zip](https://github.com/bulletphysics/bullet3/files/3309273/btsparseSDF_imagine_patch.zip)

Not related, but we use the double-precision version of Bullet (Earth-scale world). When using doubles, vector units (SSE) are not used since not implemented, however, I think it should be fairly straightforward to translate the existing SSE instructions part into AVX ones (mostly copy/paste are replace 128 in intrisics name by 256d I assume). Do you maybe have this already done somewhere ?

gjaegy, 2019-06-28:
Hi Erwin,

have you had a chance to look at the patch ? 

FYI I'm currently trying to implement the AVX version of B3_USE_SSE/B3_USE_SSE_IN_API code. I'll let you know the result.

gjaegy, 2019-07-02:
I'm actually a bit lost since there are both v2 and v3 version of the source code. Do you currently maintain both (i.e. I should update both with the AVX stuff) ?

erwincoumans, 2019-12-18:
This should be resolved by now, I modified the hash function.
https://github.com/bulletphysics/bullet3/commit/363dc8d431c4eb4c657c21273391b78f4a26229b#diff-c261ee9ba75bc6a7cd72ff982c3def1c

Thanks for the report!

closed 2019-12-18 by erwincoumans

gjaegy, 2019-12-18:
Hi Erwin,
thanks for the fix. However, looking at the patch, the issue is still present, or am I not waken up (possibly) ? Actually the code looks exactly the same to me...

The 4 bytes resulting from the padding  of the void* member when compiling in x64 is still there, and taken into account when computing the hash (and those 4 bytes contain random content). The hash should only take into account the memory used by x/y/z/w/p, and not the memory used for padding between this member. 

So, either pack the struct tightly to have no padding, or hash the structure manually by adding each single x/y/z/w/p field content to the hash (as opposed to hasing the structure using the whole memory range used, which contains padding).

Let me know if I'm not clear enough !

erwincoumans, 2019-12-18:
I added a w component that is initialized to 0, so there is no padding: it should be fixed.
Try it out!
There is also a memset 0 but it is not needed/commented out:
https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSparseSDF.h#L347

gjaegy, 2019-12-18:
All right, I didn't noticed the addition of w when looking the dif (https://github.com/bulletphysics/bullet3/commit/363dc8d431c4eb4c657c21273391b78f4a26229b#diff-c261ee9ba75bc6a7cd72ff982c3def1c).
Good idea 👍 
In the case alignment is set to 8 bytes (default when compiling as x64), x/y/z/w should be aligned on a 4-bytes boundary, and p on a 8-bytes boundary, so that should indeed work since there will be no padding.