bulletphysics/bullet3 #2290: Slow Image rendering in pybullet

Rkartik27, 2019-06-13:
This is in reference to the issue **https://github.com/bulletphysics/bullet3/issues/1545#issue-293961716**. 

I am using a Nvidia Quadro M5000M GPU (8GB) in Ubuntu 16.04 with python 3.5.2 version. I am having a problem in replicating such a high FPS (you said that you are getting a 280 FPS) with my computer. I have even tried building pybullet from the source code using "python3 setup.py install --user" from the latest source code as I heard that the source built pybullet might have GPU support.

[image]

pybullet also seems to be using the GPU in my laptop for the code shared by you in rendertest.zip (link: https://github.com/bulletphysics/bullet3/files/1690473/rendertest.zip). The screenshot I shared above shows my nvidia-smi output in which python3 is the memory (177 MB) being used by the pybullet code (please ignore the other python 2 processes; other Tensorflow codes are using the GPU). However, I am only getting an average FPS of around 20 as seen in the print output below:

pybullet build time: Jun 12 2019 17:39:14
connecting
argv[0]=--width=640
argv[1]=--height=480
startThreads creating 1 threads.
starting thread 0
started thread 0
argc=4
argv[0] = --unused
argv[1] = --width=640
argv[2] = --height=480
argv[3] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=Quadro M5000M/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 396.54
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 396.54
Vendor = NVIDIA Corporation
Renderer = Quadro M5000M/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0
MotionThreadFunc thread started
connected
ven = NVIDIA Corporation
Testing SHARED
mean: 19.56697039069384 for 100 runs

Testing SHARED w/o shadow
mean: 18.793571272315447 for 100 runs

Writing 17186 timings for thread 0
Writing 2202 timings for thread 1
Writing 47693 timings for thread 2
numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
destroy semaphore
semaphore destroyed
Thread TERMINATED
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

Any help in my case would be highly appreciated, thx

Rkartik27, 2019-07-04:
Hi, I wanted to reply asking whether anyone has any suggestion as to what to do in my case? Because without being able to exploit the GPU power available it is no feasible to perform Deep RL simulations on pybullet. 
Please suggest any solution if available, thanks!

erwincoumans, 2020-04-21:
Can't debug everyones GPU system, good luck with tracking down the issue(s).

closed 2020-04-21 by erwincoumans