bulletphysics/bullet3 #2303: Specifying spherical joint limits for Deep Mimic humanoid

wuwingw, 2019-06-20:
Hi, I'm experimenting with the Deep Mimic code and wondered if there is a way to impose limits on the spherical joints in the [humanoid.urdf](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/humanoid/humanoid.urdf) in order to match those in Jason Peng's [original model](https://github.com/xbpeng/DeepMimic/blob/master/data/characters/humanoid3d.txt).

It seems that changing the .urdf file to add a `<limit>` tag (as with the revolute joints for the elbows and knees) does not make a difference, and I'm unsure if there is a way to set limits after loadURDF has been called in the code.

Thank you (and also thanks for the work in implementing Deep Mimic in pybullet, it's been very useful!)

jhonsonlaid, 2019-12-10:
Well, I have the same problem. Could @erwincoumans help us figure this out?

erwincoumans, 2019-12-10:
There are no limits implemented for spherical joints. I've not seen an issue so far, since the policy matches the target motion usually well. You can enable self collision.

Let's leave the issue open in case me or someone else finds time to add them.

erwincoumans, 2019-12-10:
It would be easiest to add the limits to this constraint btMultiBodySphericalJointMotor.cpp:
https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Featherstone/btMultiBodySphericalJointMotor.cpp

jhonsonlaid, 2019-12-12:
The agent's right hand tries to reach the target(samll white box), and the shoulder rotates at a pose wihch human is hard to make. So I think I need to limit the shoulder's rotate range.
[image]

arpspoof, 2020-07-30:
Hello,
Is there a schedule/plan to get this feature in the near future? I believe this is really nice to have for character animation field. It'll be great to see this implemented to bring improvements to character animation projects, not just deepmimic. Thanks!

closed 2021-04-26 by erwincoumans