bulletphysics/bullet3 #2304: More instructions on how to train DeepMimic?

KaizhaoLiang, 2019-06-23:
I was able to run the backflip demo. But when I tried to run the mpi_run.py or DeepMimic_Optimizer.py, it gives me "System cann't find file" error, which seems to come from the subprocess? Any idea how to resolve this problem? Thanks.

ZhengyiLuo, 2019-06-28:
in this repo the arg file is in a different folder. Specifically remove the "args/" part in the command: 

python3 mpi_run.py --arg_file args/train_humanoid3d_spinkick_args.txt --num_workers 4

so just: 

python3 mpi_run.py --arg_file train_humanoid3d_spinkick_args.txt --num_workers 4

Training is really slow though.... submitting an issue for that.

KaizhaoLiang, 2019-06-28:
Hi, I don't think it's not finding the arg_file. The error prompt is 
cmd: mpiexec -n 4 python3 DeepMimic_Optimizer.py --arg_file train_humanoid3d_spinkick_args.txt --num_workers 4
ERROR: Error reported: failed to launch 'python3 DeepMimic_Optimizer.py --arg_file train_humanoid3d_spinkick_args.txt --num_workers 4' on DESKTOP
Error (2) The system cannot find the file specified.

I also tried "python -m pybullet_envs.deep_mimic.mpi_run --arg_file run_humanoid3d_backflip_args.txt". It gives the same error. 

But this doesn't happen when I run testrl in both ways.

Any help is appreciated, thank you.

ZhengyiLuo, 2019-06-28:
I think there is some problem with the DeepMimic_Optimizer.py path. Try run the command at the directory where DeepMimic_Optimizer.py is. 

Specifically: pybullet3/examples/pybullet/pybullet_envs/deep_mimic/

erwincoumans, 2019-11-29:
The training speed has been fixed long ago, and things should work fine at several million samples per hour dependent on the machine/number of threads. Here is the command for a machine with 12 hardware cores:
~~~~~~~~~~~~~~~~~
python3 mpi_run.py --arg_file train_humanoid3d_spinkick_args.txt --num_workers 12
~~~~~~~~~~~~~~~~~

closed 2019-11-29 by erwincoumans