bulletphysics/bullet3 #2306: open with a GUI and then stop rendering

manuel-delverme, 2019-06-24:
Hello,
I noticed my (RL) enviroment goes at 160 FPS when render('human') is active and drops to 60FPS when mode is set to None

Is there a way to open a client with gui but then close the GUI to avoid the useless overhead?

self._p = bullet_client.BulletClient(connection_mode=pybullet.GUI)

even better would be to render on GPU with no GUI but i understand it's not possible at the moment?

erwincoumans, 2019-06-24:
It depends on your environment. Are you using 'getCameraImage'? If so, the non-gui version uses a slower CPU renderer (TinyRenderer) by default. In GUI mode, it may use the OpenGL hardware rendering.
We have EGL offline rendering, which is available. See the egl examples in Bullet/examples/pybullet/examples.

manuel-delverme, 2019-06-24:
Thanks for the prompt reply,

I'm already using the opengl rendering as such:
```
        (_, _, px, _, _) = bullet_client.getCameraImage(
            width=self.render_width, height=self.render_height,
            viewMatrix=view_matrix,
            projectionMatrix=proj_matrix,
            renderer=pybullet.ER_BULLET_HARDWARE_OPENGL
        )
```

getCameraImage

closed 2019-06-24 by manuel-delverme