bulletphysics/bullet3 #2311: DeepMimic training much slower than C++ version

ZhengyiLuo, 2019-06-28:
First thanks a ton for porting! 

Is there any reason DeepMimic training code is much much slower than C++ version from the authors (overnight pybullet trained 170 iterations while C++ version trained 12k iterations). Obviously Python is much slower than C++ but isn't the underly physics simulation mostly in C++?

ZhengyiLuo, 2019-06-28:
I assumed it was the issue with enabling double precision when compiling bullet, and went on tested it out. 

Tested it out on a docker image, at some moment it went really fast, then I exited out without saving.....

Really can not reproduce the fast running behavior, so reopening this issue....

raffaello-camoriano, 2019-07-16:
> I assumed it was the issue with enabling double precision when compiling bullet, and went on tested it out.
> 
> Tested it out on a docker image, at some moment it went really fast, then I exited out without saving.....
> 
> Really can not reproduce the fast running behavior, so reopening this issue....

Hi @ZhengyiLuo, did you find a solution to reduce training time?

ZhengyiLuo, 2019-07-16:
@raffaello-camoriano Not really. Right now I just went back to the original DeepMimic repository, which is much faster to train & run.

erwincoumans, 2019-07-17:
Yes, the underlying simulation is C++, so it doesn't make much sense. It might be that multiple MPI python processes are somehow trashing some memory caches.

I can confirm that the original DeepMimic C++ version is much faster, especially with many threads.
~~~~~~~~~~~
python3 mpi_run.py --arg_file args/train_humanoid3d_spinkick_args.txt --num_workers 24

Agent 0
-------------------------------------
|       Iteration |               0 |
|       Wall_Time |         0.00056 |
|         Samples |            4105 |
|    Train_Return |            3.18 |
|     Test_Return |            3.58 |
|      State_Mean |           0.446 |
|       State_Std |             1.6 |
|       Goal_Mean |               0 |
|        Goal_Std |               0 |
|        Exp_Rate |               1 |
|       Exp_Noise |            0.05 |
|        Exp_Temp |             0.1 |
|     Critic_Loss |           0.227 |
| Critic_Stepsize |            0.01 |
|      Actor_Loss |           0.405 |
|  Actor_Stepsize |         2.5e-06 |
|       Clip_Frac |           0.699 |
|        Adv_Mean |       -9.775452 |
|         Adv_Std |        7.488633 |
-------------------------------------

~~~~~~~~~~~

Using PyBullet:
python3 mpi_run.py --arg_file train_humanoid3d_spinkick_args.txt --num_workers 24

-------------------------------------
|       Iteration |               0 |
|       Wall_Time |           0.217 |
|         Samples |            4456 |
|    Train_Return |            2.01 |
|     Test_Return |            1.92 |
|      State_Mean |           0.332 |
|       State_Std |            2.09 |
|       Goal_Mean |               0 |
|        Goal_Std |               0 |
|        Exp_Rate |               1 |
|       Exp_Noise |            0.05 |
|        Exp_Temp |             0.1 |
|     Critic_Loss |           0.122 |
| Critic_Stepsize |            0.01 |
|      Actor_Loss |            0.41 |
|  Actor_Stepsize |         2.5e-06 |
|       Clip_Frac |           0.601 |
|        Adv_Mean |        -8.36486 |
|         Adv_Std |       2.4214108 |
-------------------------------------
~~~~~~~~~~~~~~~~~~~~

So at 24 threads, the performance is about 400 times slower. I'll try to find some time to track it down at some stage.

erwincoumans, 2019-07-21:
Performance is improved in this CL, please check it out:
https://github.com/bulletphysics/bullet3/pull/2332

~~~~~~~~~~~~~~
-------------------------------------
|       Iteration |               0 |
|       Wall_Time |         0.00104 |
|         Samples |            4659 |
|    Train_Return |            4.06 |
|     Test_Return |            4.42 |
|      State_Mean |           0.112 |
|       State_Std |            1.14 |
|       Goal_Mean |               0 |
|        Goal_Std |               0 |
|        Exp_Rate |               1 |
|       Exp_Noise |            0.05 |
|        Exp_Temp |             0.1 |
|     Critic_Loss |          0.0773 |
| Critic_Stepsize |            0.01 |
|      Actor_Loss |            0.42 |
|  Actor_Stepsize |         2.5e-06 |
|       Clip_Frac |           0.613 |
|        Adv_Mean |      -5.1922436 |
|         Adv_Std |        4.844971 |
-------------------------------------

~~~~~~~~~~~~~~

qiuyuchen14, 2019-07-22:
Thanks!!@erwincoumans 
I tried to run deepmimic, but the new version says "module 'pybullet' has no attribute 'resetJointStatesMultiDof'". I think my pybullet is not updated. I installed pybullet via "pip install pybullet". How should I solve this issue?  Thanks!

erwincoumans, 2019-07-22:
You need at least version 2.5.2 which I uploaded a minute ago. Try again using
~~~~~~~~~~~~
pip3 install -U pybullet
~~~~~~~~~~~~

qiuyuchen14, 2019-07-22:
@erwincoumans It works! Thank you so much!!

erwincoumans, 2019-07-23:
Sounds good. I also left some debugging code in the release. Will update to 2.5.3 soon, and upload a trained policy for backflip. 'walk' should be fine:
~~~~~~~~~~~~~~~~~~
pip3 install -U pybullet
python3 -m pybullet_envs.deep_mimic.testrl --arg_file run_humanoid3d_walk_args.txt
~~~~~~~~~~~~~~~~~~

raffaello-camoriano, 2019-07-30:
> Performance is improved in this CL, please check it out:
> #2332
> This enables C++ stablePD control and various array versions of PyBullet APIs
> (getJointStatesMultiDof, getLinkStates etc)

Also tested on Intel Core i7-7700HQ and runs orders of magnitude faster, thank you.

```
python mpi_run.py --arg_file train_humanoid3d_spinkick_args.txt --num_workers 4
```
```
Agent 0
-------------------------------------
|       Iteration |              15 |
|       Wall_Time |           0.042 |
|         Samples |           66547 |
|    Train_Return |            2.22 |
|     Test_Return |             2.1 |
|      State_Mean |           0.495 |
|       State_Std |            1.85 |
|       Goal_Mean |               0 |
|        Goal_Std |               0 |
|        Exp_Rate |           0.999 |
|       Exp_Noise |            0.05 |
|        Exp_Temp |          0.0995 |
|     Critic_Loss |         0.00188 |
| Critic_Stepsize |            0.01 |
|      Actor_Loss |            0.26 |
|  Actor_Stepsize |         2.5e-06 |
|       Clip_Frac |           0.147 |
|        Adv_Mean |     0.013043177 |
|         Adv_Std |       0.6059316 |
-------------------------------------
```

closed 2019-08-14 by ZhengyiLuo