bulletphysics/bullet3 #2321: ExampleBrowser crashes on Ubuntu 18.04

chudur-budur, 2019-07-10:
When I try to run the `ExampleBrowser` in `bullet3/build_cmake/examples/ExampleBrowser`, it crashes. Like this:

```
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Failed to create GL 3.3 context ... using old-style GLX context
Indirect GLX rendering context obtained
Making context current
GL_VENDOR=Intel
GL_RENDERER=Intel(R) UHD Graphics 620
GL_VERSION=1.4 (4.5.0 - Build 23.20.16.5018)
GL_SHADING_LANGUAGE_VERSION=4.50 - Build 23.20.16.5018
pthread_getconcurrency()=0
Version = 1.4 (4.5.0 - Build 23.20.16.5018)
Vendor = Intel
Renderer = Intel(R) UHD Graphics 620
Aborting and deleting SharedMemoryCommon objectSegmentation fault (core dumped)
```
Why it's doing this and how to fix it?

erwincoumans, 2019-07-10:
We use it extensively on Ubuntu 18.04 without issues.

It is mostly likely an issue with the Intel drivers. Can you try upgrading the GPU drivers? Or even better, add a NVIDIA GPU with up-to-date drivers?

chudur-budur, 2019-07-11:
> We use it extensively on Ubuntu 18.04 without issues.
> It is mostly likely an issue with the Intel drivers. Can you try upgrading the GPU drivers? Or even better, add a NVIDIA GPU with up-to-date drivers?

I already have an NVIDIA GPU and all the drivers installed. But I am not sure why this bullet3 application is loading the intel drivers. Any idea?

erwincoumans, 2019-07-11:
Disconnect the monitor from the Intel GPU and reboot. We are not providing GPU/operating system issues, best to contact Intel.

closed 2019-07-11 by erwincoumans