bulletphysics/bullet3 #2328: Using getAABB to get the object L,W,H, I actually get L+0.006, W+0.006, H+0.006

jaks19, 2019-07-15:
Method 1: Say I load an .obj (same issue with .stl) mesh loaded through a URDF file in pybullet and I compute the size of the bounding box as the difference `getAABB(obj_id)[1]-getAABB(obj_id)[0]`.

Method 2: In parallel, I compute the bounding box size by iterating over the vertices of the mesh, independently of pybullet, and fetching the min and max in the x, y and z directions, and subtracting (method 2).

**The width, length and height of the object using method 1 are each a constant factor (+0.006) bigger than the same values obtained in method 2.** 

Example: 
```
# Elephant:
# pybullet: [0.05110737 0.03902986 0.02552092]
# not pybullet: [0.04510110429905434, 0.0330236562027318, 0.019515798563117363]

# Chair:
# pybullet: [0.04284145, 0.04116153, 0.03009887]
# not pybullet: [0.03684144828843615, 0.03516152963990365, 0.024098868816453588]

# Ball:
# pybullet: [0.104816, 0.127296, 0.127908]
# not pybullet: [0.098816, 0.121296, 0.121908]
```

I have tried many meshes, and I attach the [mesh](https://www.dropbox.com/sh/3pxcjbod41e1xzt/AAAGeiSx5hRA7-80SnLnHZZ-a?dl=0) which is linked to, in the hopefully-minimal URDF file below (file name should be object.urdf and mesh name elephant.obj):

```
<robot name="block">
  
  <link name="block">
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0" />
       <mass value="1" />
       <inertia ixx="1" ixy="0" ixz="0" iyy="1" iyz="0" izz="1" />
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0" />
      <geometry>
        <mesh filename="elephant.obj" name="mesh" />
      </geometry>
       <material name="red">
        <color rgba="1 0 0 1" />
      </material>
    </visual>
    <collision name="block_collision">
      <origin rpy="0 0 0" xyz="0 0 0" />
      <geometry>
        <mesh filename="elephant.obj" name="mesh" />
      </geometry>
    </collision>
  </link>

</robot>
```

Here is minimal python code for method 1:
```
import pybullet as p
p.connect(p.DIRECT)
idx = p.loadURDF(fileName='./object.urdf', basePosition=[0,0,0], baseOrientation=[0,0,0,1], useFixedBase=True, globalScaling=1.0)
boundaries = p.getAABB(idx)
lwh = np.array(boundaries[1])-np.array(boundaries[0])
print(lwh)
p.disconnect()
```
One implementation of method 2 is as follows, using pymesh. I have verified that `mesh.vertices` below matches the vertices which I see when I manually open the mesh file using a text editor.

```
def get_bounding_box_lwh(pymesh_mesh):
    vs = pymesh_mesh.vertices

    x = vs[:,0]
    y = vs[:,1]
    z = vs[:,2]

    return [max(dirn)-min(dirn) for dirn in [x, y, z]]

import pymesh
mesh = pymesh.load_mesh("./elephant.obj")
lwh = get_bounding_box_lwh(mesh)
print(lwh)
```

Could anyone suggest a reason for this difference, and is there a suggested way to deal with the difference? 

Aside: My goal is to run a simulation and obtain the point cloud of an object mesh at any time. The object is loaded as above. Before I noticed the issue above, I had resorted to getting the object position and orientation from pybullet, reproduce the same configuration of the mesh loaded outside of pybullet and just read off the vertices. Translation and orientation of the mesh works fine, but I noticed the mesh loaded outside of pybullet is a tad smaller, which led me to the above. 

Thanks for the help.

jaks19, 2019-07-15:
I edited the above as I realized it is actually a padding of +0.003 that appears around the object, and thus the overall L, W and H have +0.006.

Does this mean that to obtain my point cloud, as I mentioned at the end of the last post, I can use the position and orientation of the object from pybullet as I intended, transform my object independently of pybullet using the position and orientation, and extracting the vertices? 

Also it would be cool to know how this padding affects collisions?

Thanks

erwincoumans, 2019-07-18:
The AABB can be padded for various reasons, including some default collision margin (see gUrdfDefaultCollisionMargin in the source code). If extreme precision is needed, I would look for other solutions or start becoming more familiar with the internals of Bullet/PyBullet.

closed 2019-07-18 by erwincoumans

jaks19, 2019-07-27:
Hi Erwin. Thanks for your time.

If I wanted to change `gUrdfDefaultCollisionMargin` to a much smaller value, is there a way to do so in python?

If not, I have searched the source code and I see `gUrdfDefaultCollisionMargin` appearing in 3 places: 
examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp
examples/SharedMemory/physx/PhysXUrdfImporter.cpp 
examples/SharedMemory/physx/URDF2PhysX.cpp

Am I right in guessing that I need to change the value in BulletUrdfImporter.cpp? (And excuse my lack of expertise with c++, but after changing the value, do I need to reinstall pybullet, from the changed source, using the build instructions provided in this repo's readme?)

Thanks!

v-wewei, 2020-03-03:
@jaks19 have you change the collision margin succesfully right now ? thanks very much.