bulletphysics/bullet3 #2329: New API Status Request

ForgeMistress, 2019-07-16:
Hello! I was wondering if there were any plans to continue development on the new API.

erwincoumans, 2019-07-18:
PyBullet is our current focus and it has a C and C++ variant of the API. See 
https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3

closed 2019-07-18 by erwincoumans

ForgeMistress, 2019-07-18:
Ok. If someone were interested in implementing Bullet 3, would it be possible to talk to you about what your thoughts were in implementing it? I have looked over the Bullet 3 code, specifically the b3Cpu Pipeline as that is what would be the most interesting pipeline for my project. There is definitely something workable there in the Bullet 3 code and I have ideas about how it could be effectively multi-threaded, I would like to know your thoughts first, however.