bulletphysics/bullet3 #2331: Is there any document that illustrates the physical meaning of state and action in pybullet environments (e.g., Ant, Hopper, Walker2D))

QuXinghuaNTU, 2019-07-17:
Currently, I want to use the environments of Ant, Hopper, Walker2D and Halfcheetah.
However, it's quite hard to understand the physical meaning of each dimension in state and action space in the xml code. Does anyone know the related document that explains the detail of the state and action space?

yokian, 2020-02-12:
For a state description [this](https://github.com/bulletphysics/bullet3/blob/8504dc78167e0c9f64faf846d15d6404ced07e97/examples/pybullet/gym/pybullet_envs/robot_locomotors.py#L35) might help .

WilliamTambellini, 2020-08-08:
The actions seem to be the motor torques for all joints :
https://github.com/bulletphysics/bullet3/blob/8504dc78167e0c9f64faf846d15d6404ced07e97/examples/pybullet/gym/pybullet_envs/robot_locomotors.py#L30

closed 2021-04-26 by erwincoumans