bulletphysics/bullet3 #2333: moving with InverseKinematics and control2 creates error

nicolasCheron, 2019-07-22:
Hi, I am trying to move a universal robot from a URDF file, using the Inverse kinematics and the JointMotorControl2 functions. The robot moves to a final position, with the good joint values but the end effector is not at the expected cartesian position as the precision should be under the millimeter  . Does someone already face this issue and solve it ?
`    UR10 = path_to_urdf
    #The robot has 6 revolute links + 1 fixed link to the end effector
    p.connect(p.GUI)
    p.setAdditionalSearchPath(pybullet_data.getDataPath())
    p.loadURDF("plane.urdf",useMaximalCoordinates=True)
    robot = p.loadURDF(UR10,[0,0,0.8],[0,0,0,1],useFixedBase=1)
    nbJoints = p.getNumJoints(robot)
    ee = 6
    for i in range(nbJoints):
        p.resetJointState(robot, i, 0)
        
    targetP = [0.5,-0.50,0.5]
    targetO = [0,0,0,1]
    jointPoses = p.calculateInverseKinematics(robot,ee,targetP,targetO)
    for i in range(nbJoints-1):
        p.setJointMotorControl2(bodyIndex=robot,
            jointIndex=i,
            controlMode=p.POSITION_CONTROL,
            targetPosition=jointPoses[i],
            targetVelocity=0,
            force=500,
            positionGain=0.003,
            velocityGain=0.1)
        
    print("target position : {}".format(targetP))
    print("target joints : {}".format(jointPoses))
    for i in range(500):
        p.stepSimulation()
        time.sleep(1/240)
    finalJoint = []
    for i in range(nbJoints):
        a = p.getJointState(robot,i)
        finalJoint.append(a[0])
    print("final link world position: {}".format(p.getLinkState(robot,ee)[0]))
    print("final joints : {}".format(finalJoint))
    p.disconnect()
        
`

returned values :
target position : [0.5, -0.5, 0.5]
final link world position: (0.5325052730924337, -0.5281429890347299, 0.4795637174764034)
target joints : (-1.0016532031561431, -0.1162552680041453, 1.2549116939167753, 0.4404806391689809, -1.5707677037079932, -0.5702514917274571)
final joints : [-1.0016532034348635, -0.11625528113225829, 1.2549116938702134, 0.4404807214189218, -1.570767703692284, -0.5702514919110917, 0.0]

erwincoumans, 2019-07-22:
For reproduction we need a full working PyBullet file (including the import pybullet lines and its dependencies) and the UR10 URDF (and its assets), where is it?

nicolasCheron, 2019-07-22:
Changing the maximum number of iterations has solved my issue, thanks for your quick reply !

erwincoumans, 2019-07-22:
Mmm, the default is 20 IK iterations, how many iterations do you end up using?

nicolasCheron, 2019-07-22:
ended up using 50 iterations to get a 0.1mm accuracy.

erwincoumans, 2019-07-23:
Thanks, let's close it then.

closed 2019-07-23 by erwincoumans