bulletphysics/bullet3 #2334: What Euler angle is the used in the pybullet?

jaydenwu17, 2019-07-23:
Hi there,

I am wondering which Euler angle convention is used in the pybullet? And is the Euler angle intrinsic or extrinsic? I have attached a wiki page which describes the difference between intrinsic and extrinsic here: https://en.wikipedia.org/wiki/Euler_angles

Thanks.

erwincoumans, 2019-07-23:
In most cases, Bullet uses quaternions and matrices by default. For btRigidBody, the rotation is specified in a fixed world frame. For btMultiBody, joint angle rotations are in local frame.

closed 2019-07-23 by erwincoumans

jingxixu, 2022-02-13:
Hi, I also have the same question but I don't think you have answered it.

The difference between intrinsic and extrinsic is if you rotate around the axis of the newly generated axis or the original axis. We all know that [X, Y, Z] Euler angle means rotation along X-axis, Y-axis, and then Z-axis. However, after the first rotation, a new frame is generated. Is the second rotation along the new Y-axis (**intrinsic**) or the original Y-axis (**extrinsic**)?

hongtaowu67, 2022-02-13:
From my past experience, I believe it when you specify [a, b, c] as your Euler angle, the rotation matrix you get is Rz(a)Ry(b)Rx(c).

jingxixu, 2022-02-13:
Do you have any references? I find it annoying that people don't rigorously define the Euler angle type for their applications.

hongtaowu67, 2022-02-13:
I don't have any reference. And I didn't find a description on the Pybullet quickstart guide. I found this out by sending in different input.

jingxixu, 2022-02-13:
According to your rotation matrix, the sequence of transformation for Euler angle [a, b, c] is

1) rotate along the x-axis for c degrees
2) rotate along the y-axis for b degrees
3) rotate along the z-axis for a degrees

Also, each transformation is relevant to the newly generated frames.

Does it sound right?

hongtaowu67, 2022-02-13:
I think if is correct. But you can also view it as:

1. rotate along the z-axis for a degrees
2. rotate along the y-axis for b degrees
3. rotate along the x-axis for c degrees

with each transformation relevant to the fixed world frame. But in the end, the rotation matrix will be the same R = Rz(a)Ry(b)Rc(z)

jingxixu, 2022-02-13:
Thanks for your reply! 

This is interesting. Basically, you are saying if we look from left to right, it is intrinsic. If we look right to left, it is extrinsic. Why is that? Any references?

hongtaowu67, 2022-02-13:
Yes. You may refer to Sec. 5.3 of "Harmonic Analysis for Engineers and Applied Scientists (Updated & Expanded Edition)" by G. S. Chirikjian and A. B. Kyatkin.