bulletphysics/bullet3 #2336: 2D dynamics in PyBullet

LemonPi, 2019-07-24:
Is it possible to have 2D dynamics in PyBullet? My efforts so far haven't been promising.
I instantiated 2 balls on x-y plane and applied forces along the x-y plane, but I end up with z motion and rotation along axes other than z-axis.

I've looked into createConstraint to maybe keep the objects on the x-y plane and rotate only along z, but the API requires that I constrain all position and orientations relative to the world frame.

I saw some posts about doing this in Bullet, but is this possible in PyBullet?

erwincoumans, 2019-07-24:
See https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/biped2d_pybullet.py

closed 2019-07-24 by erwincoumans

LemonPi, 2019-07-25:
Ok, using 2 prismatic joint constraints to keep the torso planar. I applied the same method to a cylinder and a ball and got them to move only on the x-y plane, but now they are not collision checked against each other. Are links attached by joints not collision checked? Do you have any recommendations on getting collisions back? The 2 urdfs are:

(cylinder; used to be a ball)
```
<?xml version="1.0" encoding="utf-8"?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from models/ball.xacro              | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="ball" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <link name="base_link">
    <contact>
      <stiffness value="100"/>
      <damping value="0"/>
      <rolling_friction value="0"/>
      <spinning_friction value="0"/>
    </contact>
    <visual>
      <geometry>
        <cylinder length="5.2" radius="1.0"/>
      </geometry>
      <material name="blue">
        <color rgba="0 32 255 1"/>
      </material>
    </visual>
    <collision>
      <surface>
        <bounce restitution_coefficient="1.0" threshold="0.01"/>
      </surface>
      <geometry>
        <cylinder length="5.2" radius="1.0"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="2"/>
      <inertia ixx="0.8" ixy="0" ixz="0" iyy="0.8" iyz="0" izz="0.8"/>
    </inertial>
  </link>

  <link name="world">
    <inertial>
      <mass value="0"/>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <inertia ixx="0" ixy="0" ixz="0" iyy="0" iyz="0" izz="0"/>
    </inertial>
  </link>
  <link name="y_prismatic">
    <inertial>
      <mass value="0.01"/>
      <inertia ixx="0.2125" ixy="-0.005" ixz="0.0225" iyy="0.205" iyz="0.045" izz="0.0125"/>
      <origin rpy="0 0 0" xyz="0 0 0"/>
    </inertial>
  </link>
  <joint name="y_to_world" type="prismatic">
    <parent link="world"/>
    <child link="y_prismatic"/>
    <axis xyz="0 1 0"/>
    <limit effort="0.0" lower="1" upper="-1" velocity="1000.0"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
  <link name="x_prismatic">
    <inertial>
      <mass value="0.01"/>
      <inertia ixx="0.2125" ixy="-0.005" ixz="0.0225" iyy="0.205" iyz="0.045" izz="0.0125"/>
      <origin rpy="0 0 0" xyz="0 0 0"/>
    </inertial>
  </link>
  <joint name="x_to_y" type="prismatic">
    <parent link="y_prismatic"/>
    <child link="x_prismatic"/>
    <axis xyz="1 0 0"/>
    <limit effort="0.0" lower="1" upper="-1" velocity="1000.0"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
  <joint name="base_to_x" type="continuous">
    <parent link="x_prismatic"/>
    <child link="base_link"/>
    <axis xyz="0 0 1"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
</robot>
```

(ball)
```
<?xml version="1.0" encoding="utf-8"?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from models/robot.xacro             | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="point" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <link name="base_link">
    <contact>
      <stiffness value="100000000"/>
      <damping value="0"/>
      <rolling_friction value="0"/>
      <spinning_friction value="0"/>
    </contact>
    <visual>
      <geometry>
        <sphere radius="0.1"/>
      </geometry>
      <material name="orange">
        <color rgba="255 215 0 1"/>
      </material>
    </visual>
    <collision>
      <surface>
        <bounce restitution_coefficient="1.0" threshold="0.01"/>
      </surface>
      <geometry>
        <sphere radius="0.1"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="5"/>
      <inertia ixx="0.02" ixy="0" ixz="0" iyy="0.02" iyz="0" izz="0.02"/>
    </inertial>
  </link>

  <link name="world">
    <inertial>
      <mass value="0"/>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <inertia ixx="0" ixy="0" ixz="0" iyy="0" iyz="0" izz="0"/>
    </inertial>
  </link>
  <link name="y_prismatic">
    <inertial>
      <mass value="0.01"/>
      <inertia ixx="0.2125" ixy="-0.005" ixz="0.0225" iyy="0.205" iyz="0.045" izz="0.0125"/>
      <origin rpy="0 0 0" xyz="0 0 0"/>
    </inertial>
  </link>
  <joint name="y_to_world" type="prismatic">
    <parent link="world"/>
    <child link="y_prismatic"/>
    <axis xyz="0 1 0"/>
    <limit effort="0.0" lower="1" upper="-1" velocity="1000.0"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
  <link name="x_prismatic">
    <inertial>
      <mass value="0.01"/>
      <inertia ixx="0.2125" ixy="-0.005" ixz="0.0225" iyy="0.205" iyz="0.045" izz="0.0125"/>
      <origin rpy="0 0 0" xyz="0 0 0"/>
    </inertial>
  </link>
  <joint name="x_to_y" type="prismatic">
    <parent link="y_prismatic"/>
    <child link="x_prismatic"/>
    <axis xyz="1 0 0"/>
    <limit effort="0.0" lower="1" upper="-1" velocity="1000.0"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
  <joint name="base_to_x" type="continuous">
    <parent link="x_prismatic"/>
    <child link="base_link"/>
    <axis xyz="0 0 1"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
</robot>
```

LemonPi, 2019-07-26:
I suspected that the two objects were using the same world frame, but renaming all the links and joints of one object didn't help. Using `URDF_USE_SELF_COLLISION` in `loadURDF` also didn't help.

LemonPi, 2019-07-27:
Increasing the ball's stiffness and (surprisinginly?) increasing the simulation time step resolved this issue. I still don't understand why adding in these joints impacted the parameters I had to use.