bulletphysics/bullet3 #2347: Pybullet not recognizing hollowed part of hollowed cylinder OBJ file

aznshodan, 2019-08-01:
I am using a hollowed cylinder OBJ file but as you can see below, pybullet doesn't seem to recognize the hollowed part. I attached the blender file, obj files. Any suggestions would be much appreciated.

[image]

[hollowed_cylinder.zip](https://github.com/bulletphysics/bullet3/files/3455517/hollowed_cylinder.zip)

erwincoumans, 2019-08-01:
Do you want the cylinder to be dynamic moving or static environment?
For dynamic moving, you can consider using convex decomposition, using V-HACFD. There is a utility to do this in the Extras folder, use premake.
For static geometry, you can flag the obj mesh as concave: modify the URDF file and in the collision tag add 
~~~~~~~~~~~~~
<collision concave="yes">
~~~~~~~~~~~~~
Only do this when needed, otherwise it hurts performance and stability.

closed 2019-08-01 by erwincoumans

aznshodan, 2019-08-02:
I see. Thank you for the quick response!