bulletphysics/bullet3 #2349: Remove unnecessary null pointer checks

elfring, 2019-08-01:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [Planar2D::exitPhysics](https://github.com/bulletphysics/bullet3/blob/cdd56e46411527772711da5357c856a90ad9ea67/examples/Planar2D/Planar2D.cpp#L269 "Planar2D::exitPhysics function")
- [SW_And_OpenGLGuiHelper](https://github.com/bulletphysics/bullet3/blob/cdd56e46411527772711da5357c856a90ad9ea67/examples/StandaloneMain/main_sw_tinyrenderer_single_example.cpp#L81 "Destructor for SW_And_OpenGLGuiHelper")

erwincoumans, 2019-08-01:
The check doesn't harm, I'd prefer to have it there.

closed 2019-08-01 by erwincoumans

elfring, 2019-08-01:
Would any other developers like to avoid another bit of redundant condition checks?

erwincoumans, 2019-08-01:
If you put a btCollisionObject* as a ground plane in the Planar2D demo, your modification would make it crash. For this kind of things, I'd rather be on the safer side in C++.

elfring, 2019-08-01:
How do you think about to improve affected source code by the usage of smart pointers?