bulletphysics/bullet3 #2352: pybullet HumanoidBulletEnv-v0 score is not normal

zhan0903, 2019-08-04:
Hi, I use the Openai Spinningup td3 algorithm to test pybullet's HumanoidBulletEnv-v0 environment, but got the test score around 1600 even from the beginning which is not reasonable (td3 should not work in this benchmark), why is it like this?  Does anyone have similar results? Thank you.

erwincoumans, 2019-08-04:
Can you share the learning curve and a video of the resulting policy? Also, steps to reproduce it, in detail (pip installs, apt-get installs etc)

zhan0903, 2019-08-05:
> Can you share the learning curve and a video of the resulting policy? Also, steps to reproduce it, in detail (pip installs, apt-get installs etc)

Hi, I install the pybullet through `pip install pybullet`, and use the Openai td3's default setting for the test. Below is its first test result(average over ten times test) in its 5000 steps(TotalEnvInteracts) which is around 1600.
[image]
Every seed(0-4) has similar testing results and no significant change during the whole training process(the test results stay around 1600).

erwincoumans, 2019-08-05:
Do you have a video of the resulting policy?

zhan0903, 2019-08-05:
> Do you have a video of the resulting policy?
> 
> I am not familiar with Openai td3. How do you exactly install it? What version of Tensorflow, gym etc?

Hi, TensorFlow version is 1.14.0, gym version:0.14.0, I install it just flow its official instruction on its website [openai spinningup](https://spinningup.openai.com/en/latest/user/installation.html). I tried to run the resulting policy, but got an error "pybullet.error: Not connected to physics server.". To run a simulation on pybullet, is there a particular setting need?

zhan0903, 2019-08-06:
> Do you have a video of the resulting policy?
> 
> I am not familiar with Openai td3. How do you exactly install it? What version of Tensorflow, gym etc?

By the way, how to save the simulation to a video? I can run it on my computer(for humanoid env, the policy seems failed), but don't know how to save it to a video. I tried to use gym.wrappers.Monitor to save it but occurred this error "TypeError: 'module' object is not callable" .

brokenBrain, 2019-12-07:
I'm having the same issue. In my case, I'm instantiating the environment twice. The first environment seems to be behaving normally. But the second environment is misbehaving in exactly the above manner.

erwincoumans, 2019-12-18:
I haven't trained humanoid myself. Did you check out https://github.com/hill-a/stable-baselines/releases
and https://github.com/araffin/rl-baselines-zoo
It has a trained policy using ppo2 and td3.

forcecore, 2021-01-07:
This is a real issue in pybullet-gym implementation and it can be fixed : https://github.com/benelot/pybullet-gym/pull/61

closed 2021-04-26 by erwincoumans