bulletphysics/bullet3 #2358: Getting all solutions for calculateInverseKinematic

nicolasCheron, 2019-08-07:
Hello, As I understand, the calculateInverseKinematics function returned the closest articular solution for a desired cartesian position (and optionnaly orientation) from the current  articular position.
 Is there a way to get all the articular solutions that lead to the cartesian position and not only the closest ?

jrz371, 2019-08-09:
No.
For a redundant manipulator (more degrees of freedom than required) there are an infinite number of solutions.

nicolasCheron, 2019-08-09:
Actually I am using an holonomic robot (6 joints (rotations) for 6DOF). So the number of solution is not infinite. Do you know a (python) IK solver that returns all configurations using a URDF or a DH modelisation? 
Thanks for your answer.

erwincoumans, 2019-08-09:
Even with 6 joints and 6DOF you can have infinite many solutions.
PyBullet allows you to tune the solution, using various parameters, in particular using a null space. This can be used to generate many solutions.

closed 2019-08-09 by erwincoumans