bulletphysics/bullet3 #2368: Resume training from deepmimic

ked19, 2019-08-15:
When I terminate the training and get the results in out/ directory.
Is there any method to resume the training?
I put the result to pybullet_data/data/policies/humanoid3d/ and used mpi_run.py to restart it, but the iteration number always counted from 0.

AAInsanity, 2019-08-15:
I am having the same question. It seems like the episode return always gets a big drop when I save and reload a model for retraining. I wonder if that's what's supposed to happen.

erwincoumans, 2019-12-18:
I guess you'll have to restart from scratch.
Eventually I'll turn this into a proper Gym environment, so you can use your favorite training algorithm that has a working save/restore.

closed 2019-12-18 by erwincoumans