bulletphysics/bullet3 #2369: Training DeepMimic Walk Policy

AAInsanity, 2019-08-15:
I am currently trying to reproduce the humanoid3d_walk policy using the DeepMimic environment, but I for some reasons my humanoid fails to achieve natural motion (after training for a long time). From my experience, at the early stage of training (with episode reward at around 70), the humanoid will learn to mimic the reference walking motion, but can only keep balance for about 8 steps. Later (with episode reward at 150), the agent would develop a peculiar walking motion, hopping with one foot, with the other foot trailing behind to keep balance. Then (with episode reward at 250), the motion would become a little more natural, but the guy still looks crippled. 

I wonder if anybody have had some success in training the humanoid to walk, and if so how long did it take you guys to do so? I noticed that in the Deep Mimic paper it only took them 4 * 10^7 samples to converge. I'm running well beyond that number but there's still no signs of natural motion.

By the way, I also noticed that, in the original paper, Peng used a step size of 5 * 10 ^-5 for humanoid. However, in the original code, the step size is set to 2.5 * 10^-6. I tried with the latter and the progress is even slower. So I've set it to 5 * 10 ^-5 for my experiments. If you have succeeded in training with Pybullet Deepmimic, what step size did you use?

Sorry about this long post, and I would hugely appreciate any help and insight that you would like to offer!!!

Thank you very much:):)

erwincoumans, 2019-08-15:
I would use the unmodified pybullet_envs.deep_mimic code, not change the timestep.
Here is the walking gait I get after training:
https://www.youtube.com/watch?v=ZVfxgaQzBWU

~~~~~~~~~~~~
pip3 install -U --no-cache-dir pybullet
python3 -m pybullet_envs.deep_mimic.testrl --arg_file run_humanoid3d_walk_args.txt
~~~~~~~~~~~~

Note that the reward function doesn't include some components, such as reward for similar root/base position and orientation. If we implement all the parts of the reward function, it should be identical to deep_mimic.

Can you share a video?

ked19, 2019-08-20:
erwincoumans,

How many iterations did you train?
My current training output is listed as below and one leg is still trailing behind.
-------------------------------------
Iteration            4251 
Wall_Time         7.87 
Samples            17861183 
Train_Return      33.4 
Test_Return        103 
State_Mean        0.0443 
State_Std            1.02 
Goal_Mean         0 
Goal_Std             0 
Exp_Rate             0.777 
Exp_Noise           0.05 
Exp_Temp           0.0277 
Critic_Loss           0.00489 
Critic_Stepsize     0.01 
Actor_Loss           0.251 
Actor_Stepsize     2.5e-06 
Clip_Frac              0.142 
Adv_Mean           0.047412183 
Adv_Std               1.1984057 
-------------------------------------

ked19, 2019-08-20:
erwincoumans,

How many iterations did you train?
My current training output is listed below and one leg is still trailing behind.
////////////////////////////
Iteration:             4251 
Wall_Time:          7.87 
Samples:             17861183 
Train_Return:       33.4 
Test_Return:         103 
State_Mean:         0.0443 
State_Std:             1.02 
Goal_Mean:          0 
Goal_Std:              0 
Exp_Rate:              0.777 
Exp_Noise:            0.05 
Exp_Temp:            0.0277 
Critic_Loss:            0.00489 
Critic_Stepsize:      0.01 
Actor_Loss:            0.251 
Actor_Stepsize:      2.5e-06 
Clip_Frac:               0.142 
Adv_Mean:            0.047412183 
Adv_Std:               1.1984057

ked19, 2019-08-21:
After additional training for all night,
the walking policy was not improved.
How can I reproduce the success training result?
What is your environment for training?

The current state lists below.
////////////////////////////
Iteration: 12452
Wall_Time: 23.5
Samples: 53585223
Train_Return: 95.6
Test_Return: 141
State_Mean: 0.0443
State_Std: 1.02
Goal_Mean: 0
Goal_Std: 0
Exp_Rate: 0.33
Exp_Noise: 0.05
Exp_Temp: 0.00212
Critic_Loss: 0.00151
Critic_Stepsize: 0.01
Actor_Loss: 0.241
Actor_Stepsize: 2.5e-06
Clip_Frac: 0.166
Adv_Mean: -0.32381427
Adv_Std: 1.5180116

erwincoumans, 2019-12-03:
Actually I'm running in a similar issue now on one of my machines, I will track down what is happening.

erwincoumans, 2019-12-04:
Apparently, this old version trains fine, can you check it?
git checkout bb962bdb6fdbb12eaf61327fc97356d4b44eb896
https://github.com/bulletphysics/bullet3/commit/bb962bdb6fdbb12eaf61327fc97356d4b44eb896#diff-515bf02cb28321fcf4bd83fec4b51f0b

It takes about 5 hours to train here, so bisection to track down the change may take a while... ;-)

erwincoumans, 2019-12-04:
Attached is a policy trained in about 10 hours using the older version (git checkout bb962bd)
~~~~~~~~~~~~~~~~~~~~
Iteration                Wall_Time                Samples                  Train_Return             Test_Return              State_Mean               State_Std                Goal_Mean                Goal_Std                 Exp_Rate                 Exp_Noise                Exp_Temp                 Critic_Loss              Critic_Stepsize          Actor_Loss               Actor_Stepsize           Clip_Frac                Adv_Mean                 Adv_Std                  
0                        0.0013029665417141384    4523                     4.782672578992891        5.317764783236577        0.08024304205286276      0.9413613198227497       0                        0                        0.9999497875000001       0.05000000000000001      0.0999710995384927       0.05590025809578947      0.01                     0.4262679499806836       2.5e-06                  0.557373046875           -4.233176                3.2326293                
400                      0.252718717985683        1721924                  17.619081219746377       21.612787845624005       0.018594497854555966     1.0164263509511549       0                        0                        0.9784809                0.05                     0.08834912445899822      0.0035809386287891194    0.01                     0.2754016551463043       2.5e-06                  0.08938419117647059      -0.08892649              1.0683628                
800                      0.5075504030121697       3458604                  19.869559098080863       20.934394931624837       0.018594497854555966     1.0164263509511549       0                        0                        0.9567726999999999       0.049999999999999996     0.07797075687332693      0.0031299082387276403    0.01                     0.2898263751036104       2.5e-06                  0.09237132352941177      0.020087924              0.7783768                
1200                     0.7638861793279648       5205236                  22.12352040584222        24.96063584145587        0.018594497854555966     1.0164263509511549       0                        0                        0.9349406                0.05                     0.06876247499030444      0.0026598349284338635    0.01                     0.27936602963341606      2.5e-06                  0.07747395833333333      0.13800809               0.5250766                
1600                     1.0198775747087268       6949719                  19.76620439217948        24.584684153208034       0.018594497854555966     1.0164263509511549       0                        0                        0.9131336                0.05                     0.060650446160804686     0.0032402266362944777    0.01                     0.28477675067808694      2.5e-06                  0.12224264705882351      -0.09770461              0.5449888                
2000                     1.276188032163514        8697156                  23.695351731989096       28.078522238798705       0.018594497854555966     1.0164263509511549       0                        0                        0.8912920875             0.05                     0.05348478061554293      0.0029745395545938257    0.01                     0.28120022417149615      2.5e-06                  0.12369791666666666      0.21337323               0.7336402                
2400                     1.5331768587562773       10447218                 23.79504457985614        33.86766104924083        0.018594497854555966     1.0164263509511549       0                        0                        0.8694159499999999       0.05                     0.047156316619473206     0.0038226035127687887    0.01                     0.27538818115782404      2.5e-06                  0.14192708333333334      -0.090961844             0.7483715                
2800                     1.7895032044914034       12195098                 23.29233638315169        33.82644480100053        0.018594497854555966     1.0164263509511549       0                        0                        0.847566825              0.05                     0.04158311913575193      0.002446718888576369     0.01                     0.2958915551296538       2.5e-06                  0.13910590277777776      0.19802353               0.43166742               
3200                     2.0451931881904604       13937997                 24.693321616085687       34.004700829793016       0.018594497854555966     1.0164263509511549       0                        0                        0.8257810875             0.05                     0.036681975693007796     0.003852330742827487     0.01                     0.2741138792286316       2.5e-06                  0.11783854166666666      -0.05089954              0.6390814                
3600                     2.3003621696763568       15679736                 20.836012716598947       41.894945136473524       0.018594497854555966     1.0164263509511549       0                        0                        0.8040083                0.05                     0.03236091181640446      0.004648597024403417     0.01                     0.2657220370738822       2.5e-06                  0.11029411764705882      -0.161328                1.3126234                
[... cut ...]
12800                    10.37058892832862        70752480                 178.13414698130327       162.3010919339494        0.018594497854555966     1.0164263509511549       0                        0                        0.2                      0.05                     0.0010000000000000002    0.018582046981329634     0.01                     0.2542915573274648       2.5e-06                  0.22178819444444445      2.1241348                1.4441637                

~~~~~~~~~~~~~~~~~~~~
[walk_ok_output (1).zip](https://github.com/bulletphysics/bullet3/files/3919739/walk_ok_output.1.zip)

erwincoumans, 2019-12-05:
After talking to Jason Peng, sometimes the training of the walk policy gets stuck into this local maximum. I've re-run the training a few times and it get a better policy on that machine as well.

closed 2019-12-05 by erwincoumans

hiraz01, 2025-09-02:
I have setup PyBullet with the correct versions to run the pybullet implementation of Deep Mimic environment. I seem to be getting the correct results but the Wall Time seems to be at least 2 orders of magnitude slower. I am into week 2 and this is my result so far:

Iteration                Wall_Time                Samples                  Train_Return             Test_Return              State_Mean               State_Std                Goal_Mean                Goal_Std                 Exp_Rate                 Exp_Noise                Exp_Temp                 Critic_Loss              Critic_Stepsize          Actor_Loss               Actor_Stepsize           Clip_Frac                Adv_Mean                 Adv_Std                  
0                        0.13665183431572384      4130                     3.6659319030308026       4.791040662741891        0.05812168974362766      0.9748085650135472       0                        0                        0.9999488249999999       0.049999999999999996     0.09997054564061494      0.1352985117584467       0.01                     0.30878076950709027      2.5e-06                  0.39375                  -8.931331                3.3039882                
400                      61.047410462035074       1649453                  15.709405306251563       21.586906654305544       0.03360706966206881      1.0157715315149831       0                        0                        0.9793822                0.05                     0.08880869737927315      0.001979472624952905     0.01                     0.24724480230361223      2.5e-06                  0.08349609375            0.01630358               0.628668                 
800                      113.42888488981458       3295933                  16.02373050362352        24.321106413756613       0.03360706966206881      1.0157715315149831       0                        0                        0.9588012                0.05                     0.07888655668900175      0.0023963649437064305    0.01                     0.23928808327764273      2.5e-06                  0.099609375              -0.054979816             0.7019234                
1200                     175.2858325562212        4942840                  19.231830132993004       22.966492043255194       0.03360706966206881      1.0157715315149831       0                        0                        0.9382149                0.05                     0.07007082809445034      0.002198586160375271     0.01                     0.244641225785017        2.5e-06                  0.083740234375           0.027870951              0.6693228                
1600                     242.38798793362247       6590586                  23.829921969893793       25.11705775523221        0.03360706966206881      1.0157715315149831       0                        0                        0.9176182125000001       0.05000000000000001      0.062236553028822        0.0028768407719326206    0.01                     0.2388502787798643       2.5e-06                  0.107177734375           -0.0039574406            0.7687055                
2000                     291.05562954790065       8239020                  22.1633950261074         26.47359453009712        0.03360706966206881      1.0157715315149831       0                        0                        0.897012775              0.05                     0.05527540553260661      0.0029437812918331474    0.01                     0.24276702385395765      2.5e-06                  0.160400390625           -0.053939927             0.7713525                

You can see that Iteration 400 has Wall Time of 61 hours and Iteration 2000 has 291 hours. So about 10 days going now.

Hardware of Computer -
12th Gen Intel(R) Core(TM) i9-12900H (2.50 GHz)
32 GB RAM
Nvidia 3080 Ti

Any ideas on why it is so slow? Does this seem normal based on previous experience?