bulletphysics/bullet3 #2390: eglRenderer camera is static

initmaks, 2019-09-02:
Hello,

I'm having a strange issue with the `eglRenderer` plugin. The image is rendered from the same perspective every time no matter what camera position and angle settings are used. The scene is not static and has moving objects. This problem doesn't exist if eglRendered is not used.
For example, if I run **examples/testrenderer_np.py** the camera is revolving around r2d2 and images are rendered properly from different angles at every step, however, if I use **examples/testrenderer_egl.py** the camera is static.

Any ideas why it could be happening?

Thanks!

initmaks, 2019-09-04:
The latest version of `pybullet` where the issue doesn't exist is `2.4.6`.

muupan, 2019-09-13:
I have the same problem with the latest master 66fc3a9ce94cbf9793898d9ebb11dec587670ada.

muupan, 2019-09-13:
I have confirmed that reverting https://github.com/bulletphysics/bullet3/pull/2134 solves the issue.

erwincoumans, 2019-09-24:
It looks like the line 
~~~~~~~~~~~~~~~~~
m_data->m_camera.disableVRCamera();
~~~~~~~~~~~~~~~~~
causes it.
For now, let's remove it.

erwincoumans, 2019-09-24:
fixed here: https://github.com/bulletphysics/bullet3/pull/2419

closed 2019-09-25 by erwincoumans