bulletphysics/bullet3 #2392: User manual references nonexistent btGimpactTriangleMeshShape

mast4461, 2019-09-02:
The image on page 13 of the [user manual](https://github.com/bulletphysics/bullet3/blob/2.88/docs/Bullet_User_Manual.pdf) (at least versions 2.82 and 2.83) suggests to use `btGimpactTriangleMeshShape` to handle collisions for moving objects that can not be approximated as a single primitive, a convex hull of a triangle mesh, or a compound shape:

[image]

**There is [no occurrence](https://github.com/bulletphysics/bullet3/search?q=btGimpactTriangleMeshShape&unscoped_q=btGimpactTriangleMeshShape) of the word `btGimpactTriangleMeshShape`** in the current source code. There is however something called `btGImpactMeshShape`. Perhaps that would be the correct suggestion, although I'm not yet familiar enough with the source code to know that.

jrz371, 2019-09-02:
Correct, the btGImpactMeshShape is what it should be.
The btGImpactMeshShape handles dynamic concave shapes. In order for it to work correctly you'll have to ensure you register the btGImpactCollisionAlgorithm. Be aware though that the GImpact stuff does not work well with soft body simulations.

mast4461, 2019-09-02:
Thanks for the prompt reply and tips, @jrz371!
I'm trying to add support for dynamic concave mesh colliders to [ammo.js](https://github.com/kripken/ammo.js), a transpilation of Bullet to JavaScript, for a very specific use where I don't necessarily need any realtime physics simulation, only on demand collision detection between kinematic rigid bodies with (possibly) concave mesh colliders, or possibly compound shapes consisting of convex meshes.

closed 2021-04-26 by erwincoumans