bulletphysics/bullet3 #2394: Not deterministic with time

Muhammad441, 2019-09-03:
Hi,

I am trying to simulate a path for a manipulator in pybullet. I noticed that if I simulate the same action for the same configuration at different timesteps I get different results!

Say I step ten times after resetting the simulation and execute an action (using setJointMotorControl2 position control) I get a different output, as compared to stepping it twenty times before executing the action. Is there some flag I can set to fix this?

I basically want to get the same output every time I execute an action for a fixed configuration irrespective of time! Thanks!

erwincoumans, 2019-09-03:
PyBullet is fully deterministic, as long as you send the same commands. Provide a small reproduction case if not. If you step the simulation a different amount of times before applying a command, the world is in a different state and obviously proceeds in a different way.

closed 2019-09-03 by erwincoumans

Muhammad441, 2019-10-02:
Hey Erwin, sorry for the delayed response:

What I want to do is, if I execute an action with certain configuration of objects in the world, I should get the same result irrespective of the timestep I execute it at. Currently, If I execute the same action at different timesteps I get different outputs. 

For example: 

    for i in range(1500):
        p.stepSimulation()
   
     for j in range(len(joint_numbers)):
        joint_number = joint_numbers[j]
        p.resetJointState(robot_id, joint_number, joint_position[j])

     p.setJointMotorControlArray(robot_id,joint_number,p.POSITION_CONTROL,path[0])
     for i in range(500):
       p.stepSimulation()

yields a different output when compared to:

     for j in range(len(joint_numbers)):
        joint_number = joint_numbers[j]
        p.resetJointState(robot_id, joint_number, joint_position[j])

     p.setJointMotorControlArray(robot_id,joint_number,p.POSITION_CONTROL,path[0])
     for i in range(500):
       p.stepSimulation()

I just want to make the output of my actions dependent only on the configurations of the objects in the world and the robot's configuration,  not the time at which the action is executed!

Thanks!

erwincoumans, 2019-10-02:
The state (configuration) of the world and robots changes over time. Provide a full working small pybullet example and I explain why the state is different.

Muhammad441, 2019-10-03:
import pybullet as p
from time import sleep
import pybullet_data

physicsClient = p.connect(p.GUI)
p.setAdditionalSearchPath(pybullet_data.getDataPath()) #optionally

p.setGravity(0, 0, -10)
planeId = p.loadURDF("plane.urdf")
cubeStartPos = [0, 0, 1]
cubeStartOrientation = p.getQuaternionFromEuler([0, 0, 0])
robotId = p.loadURDF("r2d2.urdf", cubeStartPos, cubeStartOrientation)
cubePos, cubeOrn = p.getBasePositionAndOrientation(robotId)

joint_number = 9

p.resetJointState(robotId, joint_number, 0)

print(p.getJointState(robotId,joint_number))
p.setJointMotorControl2(bodyIndex=robotId,
jointIndex=joint_number,
controlMode=p.POSITION_CONTROL,
targetPosition=1)

print("total joints ",p.getNumJoints(robotId))
for i in range(1000):
    p.stepSimulation()

# for i in range(1000):
#     p.stepSimulation()

p.resetJointState(robotId, joint_number, 0)
p.setJointMotorControl2(bodyIndex=robotId,
jointIndex=joint_number,
controlMode=p.POSITION_CONTROL,
targetPosition=1)

for i in range(1000):
    p.stepSimulation()

print(p.getJointState(robotId,joint_number))

while(True):
    pass

Even when I uncomment those two lines, I would like to get the same output as the start, goal and time of stepping are unchanged

SherrySwift, 2023-08-23:
hello, I meet the same problem as yours. Have you figure out why it happens?