bulletphysics/bullet3 #2401: getBaseVelocity returns zeros in DeepMimic reward computation

raffaello-camoriano, 2019-09-06:
Hello @erwincoumans,

While extending imitation learning code to imitation+task I stumbled upon a possible upstream bug in the `HumanoidStablePD.getReward(self, pose)` method, related to the computation of the linear and angular velocities of the kinematic model's base.

In particular, 

https://github.com/bulletphysics/bullet3/blob/d2665d3d80b89ef36fad49a8479bd0dd85f65326/examples/pybullet/gym/pybullet_envs/deep_mimic/env/humanoid_stable_pd.py#L810

returns
```
<class 'tuple'>: ((0.0, 0.0, 0.0), (0.0, 0.0, 0.0))
```

which results in a wrong `root_vel_err`.

I also noticed that calling `getBaseVelocity` returns the correct value until the `stepSimulation` called here:

https://github.com/bulletphysics/bullet3/blob/d2665d3d80b89ef36fad49a8479bd0dd85f65326/examples/pybullet/gym/pybullet_envs/deep_mimic/env/pybullet_deep_mimic_env.py#L299

after which it starts returning all zeroes as reported above.

Note that the problem does not appear for the simulated humanoid root velocity

https://github.com/bulletphysics/bullet3/blob/d2665d3d80b89ef36fad49a8479bd0dd85f65326/examples/pybullet/gym/pybullet_envs/deep_mimic/env/humanoid_stable_pd.py#L809

Is it a known issue?

Thanks!

raffaello-camoriano, 2019-10-13:
Temporary workaround:

```python
    # linVelKin, angVelKin = self._pybullet_client.getBaseVelocity(self._kin_model)
    linVelKin = self._poseInterpolator._baseLinVel
    angVelKin = self._poseInterpolator._baseAngVel
```

erwincoumans, 2019-10-14:
Indeed, for the kinematic model it doesn't make sense to request dynamics state such as getBaseVelocity etc. Since they are kept explicitly in a different structure, just use those values.

raffaello-camoriano, 2019-10-14:
> Indeed, for the kinematic model it doesn't make sense to request dynamics state such as getBaseVelocity etc. Since they are kept explicitly in a different structure, just use those values.
> (the kinematic model likely has default constraints to drive the velocity to zero)

Great, I'll proceed as you confirmed.

I was just confused by seeing `self._pybullet_client.getBaseVelocity` called with the kinematic model as an argument.

Thank you very much for you answer.

erwincoumans, 2019-11-29:
Despite the zero reference base velocity, the training converged to a reasonable motions previously. I will re-train the motions and see if it a converges faster. Thanks for the report.

closed 2019-11-29 by erwincoumans

zhanshenlc, 2022-02-17:
Hi, but the velocities for the body parts of the kinematic character are still zeroes. Any idea on solving this?

rohit-kumar-j, 2023-05-05:
@zhanshenlc, for euler joints:

Note: this is pseudo code
```
joint velocity =  ( mocap_jnt[i].frame[j+1] - mocap_jnt[i].frame[j] ) / time_between_mocap_frames;
```
mocap_jnt[i].frame[j] = mocap position value of joint i in the mocap-data set, at frame j

time_between_mocap_frames ~= 0.0332, acccording to the mocap data.
Hope this helps