bulletphysics/bullet3 #2403: Simulation crash when using calculateinverseDynamics

Onur4229, 2019-09-09:
Hi all, 
I am currently having the issue that my simulation crashes as soon as I want to use calculateMassMatrix, or calculateInverseDynamics. There are no errors from pybullet itself. Instead, what I get is: 'Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)'
I do not know, why I am getting this error. I attached the urdf file with the meshes, I am using for the Panda robot arm. 
Note that I have added information to the urdf (such as mass and inertia to some links), in order to make sure that the error is not due to missing information and therefore, the added information are not correct. 
However, when I start simulating, the simulation crashes and I get the error mention earlier. 

[panda.zip](https://github.com/bulletphysics/bullet3/files/3590184/panda.zip)

erwincoumans, 2019-09-11:
Thanks for the report, a crash shouldn't happen. If I find time, I'll look into it.

Onur4229, 2019-09-11:
Hi Erwin,
thanks for your reply.
I just recognized that there was a part in the URDF which was duplicated (material definition).  I removed it and there were no warnings left, but this didn't change anything. 
However, I then recognized that the fingers of the panda are the reason for the crash. 
In addition, I gave numpy arrays to the method 'calculateInverseDynamics' as arguments. After removing the fingers from the URDF and sending the position and the velocities of the joints as a list instead of a numpy array, it worked out for me. Thanks a lot!

erwincoumans, 2019-09-11:
Could you share a small script that reproduces the crash (the panda.zip only contains URDF).

closed 2021-04-26 by erwincoumans